MatMoul/g810-led #2: Wrong color on single keys when setting color for key group

amorgner, 2016-06-20:
First, thanks a lot for providing a tool for Linux support for this keyboard, greatly appreciated!

When using the `-g` command to set the color for a group of keys sometimes the wrong color is set on random single keys.

The more keys you set within a group the more keys will have the wrong color. The effect gets worse on a faster computer (laptop vs. fast desktop computer) and seems to increase with the number of keys set in one operation, so it might be a timing issue. Also, using the `-p` command to set a profile for all keys, I end up with a couple of wrong key colors.

Repeating the command helps for single groups, so after a couple of attempts, the colors are correct. However, this work doesn't help when setting a complete profile with `-p`.

MatMoul, 2016-06-20:
Hello, thank you for your return.

On my old dual core, it works but I also noticed that he had yet to be a problem.

I'll test it on my Xeon workstation as soon as possible.

Tentatively, it may be possible to modify line 575 of keyboard.cpp.
By reducing the value of maxKeyValueCount to 1 or an intermediate value.

MatMoul, 2016-08-07:
Encounter the same problem on my old PC with the last kernel, I set maxKeyValueCount to 1 pending a better solution.

closed 2016-08-07 by MatMoul

amorgner, 2016-08-15:
Setting `maxKeyValueCount` to 1 doesn't help here. I've also tried changing other settings and modifying the code but no luck. I think the general problem persists.

MatMoul, 2016-08-16:
Ok, I just pushed a new version, if problem persist, look at ticket #4.

amorgner, 2016-08-16:
That seems to fix it, no issues so far. Thanks a lot!

jegade, 2016-08-16:
Thank you, now it is fix here too.