MatMoul/g810-led #4: Does not set all keys at once

hmschreck, 2016-08-09:
When using -an FFFFFF, some keys will stay the previous color; running the command multiple times will solve the problem, and it always happens in a VERY predictable way.

MatMoul, 2016-08-09:
-an is breaked and seem not possible with the current code...
At this time, I remove the no commit feature, but the code stay here for futur work.
I just have mapped -an to -a, -gn to -g and -kn to -k and removed commented text in -h.

hmschreck, 2016-08-09:
It happens regardless of an or a. Any logs I can get you that will help?

MatMoul, 2016-08-09:
It's very hard, I can not reproduce.
how is it going?
it's always the same keys that problem?
or more random?

hmschreck, 2016-08-09:
It's a general pattern, but I was mistaken that it's perfectly predictable.
About 75% of the keys will set. Run it again, 75% if the remaining keys
set. If I run the command repeatedly, it will eventually set them all. One
time took around 20 tries though.

MatMoul, 2016-08-09:
With reference to the issue #2, what's your CPU?

hmschreck, 2016-08-09:
Off hand, it's some mobile version of the i5. I'm not sure the exact clock
speed.

MatMoul, 2016-08-09:
I need test on a best PC as my old core 2 duo...
Can you test by removing the -O2 in makefile at line 2 :
CFLAGS=-Wall -std=gnu++11

hmschreck, 2016-08-09:
Doesn't seem to have helped at all, unfortunately.  Same rough patterns.  Any commands I should run?

MatMoul, 2016-08-10:
Thank you for your tests.

I will test the next week on my work PC (a xeon cpu and a new G810 wait me)

andrey-str, 2016-08-13:
Confirm on my laptop(CPU is i7-6820HQ)

johandc, 2016-08-16:
I can confirm this bug on i7-6600U, Linux 4.4.0-31-generic #50-Ubuntu x86_64. I have added the following to main.cpp that fixes the issue (when setting profiles):

```
diff --git a/src/main.cpp b/src/main.cpp
index 04d69a8..ac5eef9 100644
--- a/src/main.cpp
+++ b/src/main.cpp
@@ -5,6 +5,7 @@
 #include <algorithm>
 #include <fstream>
 #include <string>
+#include <unistd.h>
 #include "classes/Keyboard.h"
 
 using namespace std;
@@ -139,7 +140,7 @@ int loadProfile(string profileFile) {
     
     while (!file.eof()) {
       getline(file, line);
-      
+      usleep(100);
       if (line.substr(0, 3) == "var") {
         line = line.substr(4);
         ind = line.find(" ");
```

MatMoul, 2016-08-16:
I just tested on my workstation and I also have the problem.
I fixed the code with "usleep" in Keyboard.cpp
I also increased the variable "maxKeyValueCount" to 12.
Now, it works on my Xeon CPU.
Without negative feedback from you, I will close the tiquet

MatMoul, 2016-08-16:
following return on #2, I close the ticket

MatMoul, 2016-08-16:
And thank you to johandc for the usleep

closed 2016-08-16 by MatMoul

MatMoul, 2016-08-16:
I just reactivated no commit feature :
-an
-gn
-kn
-c
All work very well !