MatMoul/g810-led #7: Basic G610 support

fgblomqvist, 2016-08-23:
Hey, just found your code and was so happy, since it's a stepping stone on getting it done for G610.

I'm willing to provide you with any kind of dumps, testing etc. if you just tell me what you need. I am also willing to look into it myself, if you maybe can give me a pointer as to which dumps to generate (which I then can analyze). I'm trying to at least set the mode on it, since it starts out with this annoying wave that I don't want to manually change from (brightness key + 0) every time I start the computer.

As it is right now, I don't think the program is detecting the keyboard at all, no errors and no output. I'm guessing there's a minor difference in some ID somewhere. Let me know what you think.

MatMoul, 2016-08-23:
Hi, my code is mainly done for independent RGB buttons, the G610 does not seem to be part of this category.

You can always try to add a line in the main.cpp after line 214:
if (str == "G610-led") lg_pid 0xc = ???; // The pid of your keyboard.

And don't forget to rename the binary to G610-led.

fgblomqvist, 2016-08-25:
Hi, yeah no it's not, but I thought that maybe they'll have a similar way of communicating since they are in the same product line. I changed the PID to my keyboards PID (0xc333), but still nothing. I tried generating a dump with Wireshark in Windows, while setting the different light modes, but Logitechs software is being very chatty with the keyboard, so it's hard to distinguish which packages I caused. I looked at your dumps and while they are similar, there are still (as expected) minor differences. Not sure how you managed to isolate them. Do you possibly have any pointers as to what I should be looking for?

MatMoul, 2016-08-25:
Regarding the dumps, look at set_keys_colors.pcapng ...
In this capture, the interesting messages are sent to 3.2.0 ...
Adds a filter in Wireshark: usb.dst == "3.2.0"
Normally this filter commands sent.
After simply retrieve Data Framgent in the URB setup section.
For the tests, I advise you to use the version of Python
I hope this permettes you progress.

fgblomqvist, 2016-08-30:
Ah okay, thank you for the advice!

MatMoul, 2016-08-30:
I close issue but you can ask if you have others questions

shakeme, 2016-09-29:
I created a simple script based on https://github.com/MatMoul/g810-led-python.
You can find it here: https://github.com/shakeme/logitech-g610-python/blob/master/g610.py

Thank you for all the work MatMoul. You made my day. :)

MatMoul, 2016-09-29:
It's cool to see that my code can be used for another project ...
Long life to open-source...

achaios, 2016-11-09:
The program works for G610 by adding this line
else if (str == "g610-led") lg_pid=0xc333;
The first 2 digits of the colour are taken into account, however all 6 are required to work
e.g.:  g610_led -g logo 80FFFF

MatMoul, 2016-11-10:
@achaios Thank you for your return, I will try to include your change in a next release

MatMoul, 2016-11-12:
Could you try the last version, I just include changes to support g610...
It's no more required to rename the binary...

achaios, 2016-11-12:
Yep, I saw what you did there. It works. It is good that we don't need a properly named binary.
It is still necessary to specify 3 bytes for the colours, when only the first is effective. No big deal, but perhaps colours that are omitted in the command line could just get a default value, like zero? 
Thank you.

MatMoul, 2016-11-18:
Just added your request.
Sample : g810-led -a 60

achaios, 2016-11-19:
Perfect. This issue can be closed.

MatMoul, 2016-11-19:
Thank for your work

closed 2016-11-19 by MatMoul