MatMoul/g810-led #8: G910 support

jakefb, 2016-09-05:
Hi is it possible to get this to work with the G910?

MatMoul, 2016-09-05:
I think that is possible...
For testing, I suggest you replace the pid at line 12 of file src/main.cpp.
You should find the pid of the G910 with lsusb cmd.
If it works, it only remains to add the missing keys.

dedalqq, 2016-09-07:
i find...

```
      Device Name: Gaming Keyboard **G810**
         Attached: 1
    Controller ID: 0x0
Vendor/Product ID: 0x046D, **0xC331**
        FF Driver: n/a
```
```
Device Name: Gaming Keyboard **G910**
Attached: 1
Controller ID: 0x0
Vendor/Product ID: 0x046D, **0xC32B**
FF Driver: n/a
```

but i tried use 0xC32B and this is not work = (

dedalqq, 2016-09-07:
Official response from logitech support:
```
The G410, G810, and G910 should all be the same.
```
it should work. but i do not can to force to work =(

MatMoul, 2016-09-07:
I just saw that there were 2 models:
  - Logitech G910 Orion Spectrum
  - Logitech G910 Orion Spark
What is your model (give me the product manufacturer number)?
I can order one by my company for testing ;-)

dedalqq, 2016-09-07:
apparently Orion Spark....

[image]

MatMoul, 2016-09-07:
I can order a G910 Spark but Spectrum is the last version. (Is a good idead?)
The Spark seems to use a oldest protocol...
But if you can, give me a wireshark dump or lookup at issue #7...

dedalqq, 2016-09-09:
Don't mind your idea, appreciate your work already :)
Will try to work with dump and your code. Will send you dump if i can't figure it out. Or i can just send it anyway if you're interested.

MatMoul, 2016-09-09:
If you can send me your dump, I could probably help you.
During my research, I launched a Windows 7 in a WM with VirtualBox.
The Wireshark is installed on my Linux.
Before capturing the USB port, I connect the keyboard directly to the VM.
And I use the Logitech software to set all keys to a defined color.
This gives me the dump.

MatMoul, 2016-09-18:
I close issue but if you have any question, you can ask in this issue

andreast1990, 2016-11-27:
G910 Dump Test

Win 7 with VirtualBox

[G910_Test.zip](https://github.com/MatMoul/g810-led/files/614690/G910_Test.zip)

MatMoul, 2016-11-27:
@andreast1990 Can you test the branch spark
At this time, only keys from group "keys" wil work

andreast1990, 2016-11-27:
I test it, no Key change the color.

MatMoul, 2016-11-27:
No luck...
When I have time, I will extract all the data in the dump.
We need to be test each data with the python version (it's more easy than C++)

andreast1990, 2016-11-27:
It works !!!!

3 Keys don´t work (G1, G2, G3)

`bool Keyboard::commit() {
  if (m_isAttached == false) return false;
  bool retval = false;
  unsigned char *data = new unsigned char[20];
  data[0] = 0x11;
  data[1] = 0xff;
  data[2] = 0x0f;
  data[3] = 0x5d;
  for(int i = 4; i < 20; i++) data[i] = 0x00;
  retval = sendDataInternal(data, 20);
  delete[] data;
  return retval;
}`

`    case KeyAddressGroup::keys:
      data[0] = 0x12;  // Base address
      data[1] = 0xff;  // Base address
      data[2] = 0x0f;  // Base address
      data[3] = 0x3d;  // Base address
      data[4] = 0x00;  // Base address
      data[5] = 0x01;  // Base address
      data[6] = 0x00;  // Base address
      data[7] = 0x0e;  // Base address
      break;
    default:
      return false;
      break;
  }
  return true;
}`
[classes.zip](https://github.com/MatMoul/g810-led/files/614863/classes.zip)

andreast1990, 2016-11-27:
All Keys work!

Logo2 G910 not work.

[g910-led.zip](https://github.com/MatMoul/g810-led/files/615007/g910-led.zip)

MatMoul, 2016-11-27:
Good job !
But where are multimedia keys ? G910 seem to have multimedia keys ?
For the Logo, I have my idea, in the Keyboard::setKeys function, logo don't support more than one value.
I will include change in the code to support this protocol.
I start a new branch for this and I come back to you for merge and test your change.

MatMoul, 2016-11-28:
You can test the new spark branch.
I have added logo2 for your second logo.
But multimedia keys and m keys are not handled.

andreast1990, 2016-11-28:
Logo2 not work, i think Logo2 have a other base, i make a new dump off this LED.

Multimedia and M Keys always the same color, no RGB Leds.

andreast1990, 2016-11-28:
[logo_g910.zip](https://github.com/MatMoul/g810-led/files/616972/logo_g910.zip)
No. 701 - 817

MatMoul, 2016-11-28:
Thanks and OK for multimedia and m keys...
In your dump, it seem that logo to be at the line 781 and the second logo and color is here.
I search an error in my code.

MatMoul, 2016-11-28:
A possible way :
In Keyboard::populateAddressGroupInternal, data[3] = 0x3a but in dump data[3] = 0x3c...
Line 512 of the last commit

andreast1990, 2016-11-29:
In my dump i only chance the color of logo2.
Logo works with 0x3a and 0x3c.
I think logo2 have 2 LEDs.
Yesterday i tested all bases in the dump

andreast1990, 2016-11-29:
logo2 hat the same base as logo1, key is 0x02

g810-led -k logo2 ff00ff works (i not testet this yesterday)
g810-led -a ff00ff not work (error in -a funktion?)

MatMoul, 2016-11-29:
Found error in code try the last version...

andreast1990, 2016-11-30:
It works.

All Keys and Logos

MatMoul, 2016-11-30:
@andreast1990 Thanks for your work
I just merged spark branch to master

closed 2016-11-30 by MatMoul