MatMoul/g810-led #9: Ability to set speed of effect

ShadauxCat, 2016-09-14:
I like the rainbow effect, but I find its default speed to be distracting. With the windows driver I can set it to be very slow. but I can't do that with this as far as I can tell. Would be great to be able to set effect speed.

MatMoul, 2016-09-18:
I tag to enhancement...
But for now, I don't have time for this...
Wait and see...

MatMoul, 2016-12-13:
@noisycat has pushed dump for effect...
It's a first step

noisycat, 2016-12-13:
@MatMoul  - Did you have a good way to diff the pcapngs? Because at that point, i feel like its maybe a Saturday's worth of work to get this general feature in.

MatMoul, 2016-12-13:
I don't think we can do anything.
By testing on Windows, when I quit the app, the effects no longer work.
I thought they were related to the firmware effects but not.
For effects, you need a daemon and it's not what this projet is.

@noisycat : It seem that new dump are not the good, don't work on it.

noisycat, 2016-12-13:
I disagree because when you do the effects on linux it sticks around - the windows app is really good about making sure to clear the keyboard buffers / settings on full exit, so I’m not surprised it disappeared. If you want to try again, you can force quit the app and that should do it - IIRC what the forums said.

-Charles

MatMoul, 2016-12-13:
Second test :
Set effects in a WM on my Linux and disconnect the keyboard from the VM.
Effects stop working and need to go to CTRL+ALT+F2 to recover my session.

noisycat, 2016-12-13:
Third Test:
1) Set effects using standard Windows Logitech gaming software
2) standard quit
3) lose effect

Fourth test confirms what I said earlier:
1) Set color cycle effect with fast speed on standard windows logitech gaming software
2) force quit the Logitech KHAAL(sp?) setevent manager process and logitech gaming software through task manager
3) keep color effect!
4) reboot into linux-3.10 (EL7)
5) type up this report while drinking a diet coke and basking in color cycling light. Life is good.

@ShadauxCat that's a temporary work around for you if you don't want to wait for us to implement it - set it in windows, force quit out of the event manager and the gaming software, and you should be good to restart. I should note that I never got the systemd / udev stuff working, so if the above process doesn't work, you're likely just wiping it with the initial startup effect from the g810-led program

MatMoul, 2016-12-13:
@noisycat : Thank for your test but can you test with the star effect ?

noisycat, 2016-12-13:
I can confirm that your results with the star effect were not wrong!
1) star effect
2) key press effect
both require the gaming software to be running in the background (acting as a daemon as MatMoul hypothesized). However, that still means that:

1) Fixed Color
2) Breathing
3) Color Cycle
4) Color Wave

Could all be done by g810-led

ShadauxCat, 2016-12-13:
My main concern is the rainbow effect, because it's just too fast and it's distracting at its default speed. At least as far as my bug is concerned, adding support for breathing and color cycle and star and key press aren't particularly important to me; I only care really about tuning the speed of the currently supported effect (rainbow)

MatMoul, 2016-12-13:
Ok, it seem that some effects work and other not...
Need new dump for 3 to 6...

@ShadauxCat : Your request is part of effects and we work to know how it work...
It seem it have a possibility for your need ;-)

@noisycat For dump, packages size need to be 84 or 128...

noisycat, 2016-12-13:
@ShadauxCat rainbow is actually called Color Cycle by logitech - sorry if that wasn't immediately familiar. So what I'm saying is that you can achieve what you need / want _right now_ if you have access to the windows side. I do agree that the slow color change is particularly mesmerizing - especially when you have it blossoming from the center to the sides instead of wiping horizontally!

@MatMoul  I'm not sure what you mean by 84 or 128... are you referring to the entire message? the leftover data? I'm super new to using wireshark / sniffing USB in general, so you're going to have to be more explicit. I just know how to design experiments where its

1) easy to find values you set
2) easy to ensure differences between two sets are only things you don't set

ShadauxCat, 2016-12-13:
I know. All I'm saying is, if some of them won't work, that's fine. It doesn't have to be all of them. Just the ones that have been identified as working are sufficient to address the issue IMO.

MatMoul, 2016-12-13:
@noisycat : If you open one of my dump, you can add the next filter :
usb.dst == "3.2.0"
With this, you have all packet sent to the keyboard (3.2.0 is my usb port, it can be different on your config)
If you look at length column the number is 84 or 128
An other good filter is :
usb.data_fragment contains 00:dc:ff
where 00:dc:ff is the color set in the Logitech software

noisycat, 2016-12-14:
Ah - I see - well, I'm just going to get the entire dumps for the usb port like I have been doing and you can apply whatever filters you want. It takes long enough for me to setup the various profiles so that they don't have the continual communication from the LGS 8.88 spewing out additional results.

guyhalestorm, 2016-12-16:
First, y'all are my heroes.  I suck at programming and I was absolutely delighted to see someone finally made this work in Linux.  Thank you so much!

Secondly, I adapted [this guy](http://www.gossamer-threads.com/lists/python/python/819538)'s Python script to make a nice, smooth, slow color cycle effect.  It's brute force as hell, not exactly the same color cycle, and it uses some CPU, but damn it's pretty.  Might tide interested parties over until the built in color cycle effect speed settings can be changed.  :)

```
import colorsys
import subprocess

sat = 1
value = 1
length = 1000
for h in range(0, length + 1):
	hue = h / float(length)
	color = list(colorsys.hsv_to_rgb(hue, sat, value))
	for x in range(3):
		color[x] = int(color[x] * 255)
	hexval = ("%02x%02x%02x" % tuple(color)).upper()
	exitcode = subprocess.call(('/usr/bin/g810-led', '-a', '{}'.format(hexval)))
```

Enjoy!

MatMoul, 2016-12-16:
Excellent result ...
This reminds me of a old test of effects that I did in bash with the same type of solution.
But it's necessary to optimize the code for effects...
Current idea is to add timeout(time) and loop keyword in profile file

MatMoul, 2016-12-28:
I just tested the new version of Windows software and there is a new category of animation.
It's a bit like what I wanted to do (chain the profiles).
But to operate, it is necessary to use a deamon.
This project will remain a simple command line.
But now we have to find a new name for a project that will work as a deamon.
Look at this issue : Next project name : #29

Sirsyorrz, 2016-12-30:
@guyhalestorm I ran the script using ./script_name.py and nothing worked? I have the g810 and i did change it to go to /usr/bin/g810-led but it didnt do any rainbow effect or anything. Could I have some help setting it up as im new to linux and know nothing of python.

Thanks and can't wait for this to be updated :)

guyhalestorm, 2016-12-30:
@Sirsyorrz Run it using the command "sudo python script_name.py" and it should work.  I needed to run it with elevated privileges for it to work right on my machine.

Sirsyorrz, 2016-12-30:
@guyhalestorm Thanks it worked :D

Sirsyorrz, 2016-12-30:
@guyhalestorm just another question, how do i turn it off and restart it? I closed the terminal and now its just stuck and i cant do any g810-led commands at all.

ShadauxCat, 2016-12-30:
I made some modifications to the g810 tool so that you could specify `-k` multiple times and it would group them up and send them more efficiently, then added some additional effects to the script. I kind of hacked in the C++ code; if you're interested in that change I'll clean it up a bit and open a PR for it. (I made that change because having to call the app multiple times for each `-k` command was REALLY slow, batching it up into a single call was necessary for the effects to look any good at all.)

MatMoul, 2016-12-30:
Just pushed a new version with possibility to set 2 natives effect...
g810-led -fx-breathing 00ff00 10
g810-led -fx-cycle 10
wave effect comming soon...

MatMoul, 2016-12-30:
@ShadauxCat I work on a solution for this but the idea is to pipe a profile style content.

MatMoul, 2016-12-30:
wave effect working...
g810-led -fx-hwave 10
g810-led -fx-vwave 10
g810-led -fx-cwave 10

MatMoul, 2016-12-30:
In reply to the firs question, all you need is the next command :
g810 -fx-hwave 10
where 10 is the speed of the effect...
To use on boot, you can make a profile (sample in sample_profiles/fx-hwave

closed 2016-12-30 by MatMoul

ShadauxCat, 2017-01-02:
I don't see any noticeable difference between `-fx-hwave 10` and `-fx-hwave 100` and `-fx-hwave 1`, although `-fx-hwave 0` is a nightmarish epilepsy-inducing strobe. What's the range?

MatMoul, 2017-01-02:
You need to use hexa for the speed (00 to FF)

ShadauxCat, 2017-01-02:
Ah, that makes more sense. Thanks. Maybe make a note of that in the readme. :)