MatMoul/g810-led #10: My g810 has a different product ID.

ShadauxCat, 2016-10-07:
Took me a while to figure out why this wasn't working. My g810 is:

`Bus 002 Device 019: ID 046d:c337 Logitech, Inc.`

The `0x046d` for vendor ID is correct, but my product ID is `0xc337`. Seems like there's multiple types of g810 out there and this should probably try connecting to all known g810 pids.

MatMoul, 2016-10-07:
Is the script working when you change the product ID ?
It's possible to add thid product ID in the code but I don't have possibility to test.

ShadauxCat, 2016-10-07:
It is working when I change the ID. I'll see if I have time this weekend to do some research and see if I can find what other product IDs this keyboard has, and try to open a PR.

MatMoul, 2016-10-07:
OK, but now, it's not very easy to include a clean solution (thank you logitech).
I will try to work on this the next weeks.

MatMoul, 2016-10-29:
Just added multiple pid support

closed 2016-10-29 by MatMoul