MatMoul/g810-led #12: Key names for different keyboards are inconsistent

ShadauxCat, 2016-10-07:
On my keyboard, I found the following:

`-` changes the `/` key.
`?` changes the `-` key
`unknown` changes the `\` key
`'` changes nothing
`à` changes the `'` key
`§` changes the `~` key
`tidle` [misspelled, see my other issue] changes the `=` key

My guess is that different pids (see my other other issue) refer to different keyboard layouts, and you could probably use the pid that successfully connected to determine which named keys should refer to which key IDs. But since I only have one keyboard, I can't verify that.

My keyboard is US QWERTY layout for the record.

MatMoul, 2016-10-07:
Thank you for return
I created this code with a QWERTZ keyboard (fr-ch-latin1)
This is certainly the reason
From my perspective, it would be better to rely on the QWERTY (en-us)
I'll try to fix it but I would need to return ...

meisanerd, 2016-10-24:
It might be more complicated than just checking layout by USB ID, as my keyboard is also the US QWERTY layout, but the ID is 046d:c331

ShadauxCat, 2016-10-24:
Hm, well that's unpleasant. Still hopefully everything with c331 is US QWERTY so we could still use those IDs? But it's gonna be really hard unless we can get a full list of logitech product IDs.

MatMoul, 2016-10-24:
And my QWERTZ CH keyboard is also 046d:c331

derblub, 2016-10-25:
While we're at it:
German QWERRTZ  with `046d:c331` too.

meisanerd, 2016-10-25:
Well this is going to be fun...

I am not sure if this is all of the layouts, but they sell spare keycaps for the following:
US
Swiss
German
Nordic
Spanish
French
UK
Turkish
(http://support.logitech.com/en_gb/product/810-orion-spectrum-rgb-gaming-keyboard, related spare parts)

We have already established that c331 covers at least US, French?, and German, and c337 is US.

I am just thinking we might have to just add a couple more flags to the command line, one for layout, and one for USB ID (defaulting to c331 as it appears to be the most common, unless we can detect this from code).

MatMoul, 2016-10-25:
Thank you for your feedback.
I try to work on it this weekend.

MatMoul, 2016-10-29:
Just open a new branch "keynames" to work on this
I just added a -lk arg to list supported keys.
But I show than 13 keys is very difficult to maintain.
Firstly, I will align keynames to standard constante names

MatMoul, 2016-10-29:
A question for owners of a QWERTY keyboard: do the two letters y and z are reversed?

ShadauxCat, 2016-10-29:
A picture's worth a thousand words, so it's probably easiest to answer by saying our keyboards look like this: http://gaming.logitech.com/3D/12199-G810/images/keyboard_en_nologo.png

MatMoul, 2016-10-29:
Thank for the link, I just found a link with terminology desc : [http://ell.stackexchange.com/questions/84624/what-is-the-name-for-these-keys-on-a-computer-keyboard](http://ell.stackexchange.com/questions/84624/what-is-the-name-for-these-keys-on-a-computer-keyboard)

MatMoul, 2016-10-29:
In branch "keynames", I just remapped keys and need return if keys are right for a QWERTY EN-US keyboard :
tidle
minus
equal
open_bracket
close_bracket
backslash
semicolon
quote
comma
period
slash

meisanerd, 2016-10-31:
Yes, those key names are correct.

If they don't match up for some reason, I can probably submit a PR with the corrections.

MatMoul, 2016-10-31:
Thank for confirmation, branch merged

closed 2016-10-31 by MatMoul