MatMoul/g810-led #14: `-g keys` has inconsistent behavior

flarn2006, 2016-10-10:
I loaded [this sample profile](https://raw.githubusercontent.com/MatMoul/g810-led/master/sample_profiles/group_keys) and noticed that my U key was, for some reason, orange. Looking at the profile, I didn't see any references to specific keys, or even that color. So I set all the keys to the same color and tried running each line one by one. When I got to `g810-led -g keys 009696`, I noticed my 3 key had turned the same shade of orange. Running the command a few times changed the colors of other random keys, either turning the LED's off, or making them that orange color.

**To reproduce:** Simply run `g810-led -g keys 009696`, and look for any keys that aren't the color they should be. Try running it a few more times and watching for changes. Or you can even do `while true; do g810-led -g keys 009696; done` and watch random keys flicker on and off.

Azphreal, 2016-10-18:
I think it might have to do setting multiple keys at once, as (for me) loading a profile causes the same effect, even if I'm not using a group command. It's fairly consistently my 'm' key, and occasionally my 'a' or left ALT.

MatMoul, 2016-10-19:
it seem the same old problem as #2 and #4 ...
Try to increase the usleep value on of line 391 of src/classes/Keyboard.cpp

MatMoul, 2016-10-29:
Does the modification of usleep changes anything ?

flarn2006, 2016-10-30:
@MatMoul: Right now the issue isn't occurring for me even on the original version I compiled, which hasn't changed since the issue first occurred. Seems like it's intermittent.

flarn2006, 2016-11-01:
@MatMoul: Disregard my last comment; I was testing it incorrectly, using `-a` instead of `-g keys`. It still occurs even after increasing the delay to 1000000.

MatMoul, 2016-11-02:
Thank you for your test, I just tested -g keys and now, I encouter the same problem.
Other groups seem not afected.

MatMoul, 2016-11-03:
could you test by changing the value maxKeyValueCount to 1 on line 569 of keyboard.cpp.
It's just a possible temporary patch.
On my comput, with this modification, I need 300ms to update keyboard vs 150ms.

flarn2006, 2016-11-04:
@MatMoul: It partially fixes it. I don't get random keys turning orange anymore, but sometimes keys that aren't in the group change to the color I specified.

MatMoul, 2016-11-10:
@flarn2006 Thank you for your test.
I will try to find a solution but I need time.

MatMoul, 2016-12-06:
**Special thanks** to @MohamadSaada ...
He found the bug...
#20

closed 2016-12-06 by MatMoul