MatMoul/g810-led #17: Installation Failure

danwspark, 2016-11-30:
Installation documentation is a bit sparse. I can't seem to get it to work on my machine.
libusb, g++ are installed, and the g810 'makes' properly.. however the keyboard does not leave the default rainbow effect.

MatMoul, 2016-11-30:
If you don't added udev rules, you need use sudo.
Otherwise can you give me the result of
lsusb -d 046d: -v

danwspark, 2016-12-04:
[lsusboutput.txt](https://github.com/MatMoul/g810-led/files/629287/lsusboutput.txt)

let's see if attaching like this works.
I've done all of the make with sudo command as well..

hey, I really appreciate your responsiveness!

durairanubis, 2016-12-04:
I'm having a failure also: My error is:
g++ -0 bin/g810-led -Wall -02 -std=gnu++11 src/main.cpp src/classes/Keyboard.cpp -lusb-1.0
In file included from src/main.cpp:4:0:
src classes/Keyboard.h:5:44: fatal error: /usr/include/libusb-1.0/libus b.h: No such file directory compilation terminated.
IN file included from src/classes/Keyboard.cpp:1:0: src/classes/Keyboard.h:5:44: fatal error: /usr/include/libusb-1.0/libus b.h: No such file or directory compilation terminated
makefile:11: recipe for target 'bin/g810-led' failed
make: *** [bin/g810-led] error 1

Sorry I don't know how to save the terminal screen as a document yet.

MatMoul, 2016-12-04:
@danwspark :

The first line of "lsusb" returns "Could not open device, some information will be missing"
It appears that access to the USB port is not fully authorized.

What is your Linux distribution?
Have you tested as root?

Have you added the file "udev / g810.rules" to "/etc/udev/rules.d"?
This operation need a reboot

MatMoul, 2016-12-04:
@durairanubis :

It seems that "lsusb" library is not found
Have you installed the "lsusb" package?
If you do not know the name of the package, give me the distribution you use.

durairanubis, 2016-12-04:
I currently have Ubuntu 16.04.1 LTS, I just installed it and new to Linux
so its quite possible I'm missing something.

I do not know what the isusb package is.

MatMoul, 2016-12-04:
@durairanubis 
Sorry bad package name, it's "libusb"...
For ubuntu, it seem you need to install "libusb-1.0-0-dev" package
`sudo apt-get install libusb-1.0-0-dev`

danwspark, 2016-12-04:
hey @MatMoul, doing sudo g810-led <operation> worked perfectly for me. I can't believe that I just didn't think of that. Thanks a bunch! do you have a paypal anywhere? I'd like to buy you a beer if you'll let me :)

@durairanubis, try `sudo apt install libusb-1.0-0-dev` before doing install.

danwspark, 2016-12-04:
@durairanubis also for future reference, if you want to capture the output of some command line operation, do `<operation> > <filename>`

the > carot feeds the output of the operation into the filename you designate.

MatMoul, 2016-12-04:
Thank for the beer @danwspark ...
This issue stay open for @durairanubis.

MatMoul, 2016-12-11:
No return, closing

closed 2016-12-11 by MatMoul