MatMoul/g810-led #19: Error loading profile

MaxSor, 2016-12-03:
Hi, i'm happy that somebody add such support of Logitech keyboards to linux.
But i have problem with my profile.
I extract my profile from windows into xml file and place it in /etc/g810/ filename is "profile".
Then i run 
``sudo g810-led -p /etc/g810/profile``
And got error on every line of file:
```
Error on line 1 : <?xml version="1.0" encoding="utf-8"?>
Error on line 2 : <profiles xmlns="http://www.logitech.com/Cassandra/2010.7/Profile">
...
```
What is wrong? Or i'm doesn't understand smthng?

MatMoul, 2016-12-03:
Hello, at this time, linux profile as his own file format...
Look sample at [https://github.com/MatMoul/g810-led/tree/master/sample_profiles](https://github.com/MatMoul/g810-led/tree/master/sample_profiles)

MaxSor, 2016-12-04:
Sorry for my inattentiveness. Thx, this works!

closed 2016-12-04 by MaxSor