MatMoul/g810-led #21: Does this work on Windows/Cygwin/MinGW?

abhijeet080808, 2016-12-11:
Can this be made to work in Windows/Cygwin/MinGW? It would be very handy to replace the default software that logitech bundles. I am willing to offer any help that is needed.

I think if the PyUSB library can be made to work with the windows system, rest should work out easily.

MatMoul, 2016-12-11:
I just tested in Cygwin.
I can make the binary but libusb has not R/W access to the keyboard.
This feature seem limited by Windows design.

abhijeet080808, 2016-12-12:
Yes, I saw similar result with your python script too. However, both works correctly when run on a Linux VM and can actually control the keyboard that way. Maybe something to ponder upon.

MatMoul, 2016-12-12:
Sorry but no time for this, Windows need to use other api with USB HID devices.
I close issue but if anyone provide a solution, I will merge it.

closed 2016-12-12 by MatMoul