MatMoul/g810-led #26: Handling command line inputs with ASCII values > 128

noisycat, 2016-12-14:
Need to decide on how we're going to do that if we want people to be able to use even the unsigned char set, like æ as @larsnaesbye pointed out. I don't have much experience with unsigned char (or unicode) on the command line as arguments for C programs, but being able to handle it at all is a prereq of handling local character sets.

larsnaesbye, 2016-12-14:
I could live with using a standard moniker instead, like 'aelig'. Everything is better than having to remember it's the semicolon or whatever.

MatMoul, 2016-12-14:
Just use a \ before the char...
for tilde, you can use : g810-led -k \~ ff0000
always with tilde it's possible to add a second possibility as ex : g810-led -k tilde ff0000

noisycat, 2016-12-15:
tilde? He's talking about ASCII+231, not ASCII+126 (reference: http://www.theasciicode.com.ar/extended-ascii-code/latin-diphthong-ae-uppercase-ascii-code-146.html )

+231 is outside of the (char* argv[]) input that main gets. And instead of just using complements, since most of those special characters aren't in the ASCII table, it starts making sense to have the discussion.

MatMoul, 2016-12-15:
I have used the tilde sample to show that you can have multiple definition of a key
I just tested to add æ in Keyboard::parseKey and it work from command line without \

MatMoul, 2016-12-24:
Can I close this issue ?

larsnaesbye, 2016-12-25:
Sure, if it works I see no reason not to close it :-)

closed 2016-12-26 by MatMoul