MatMoul/g810-led #28: Group: gkeys

sebschlicht, 2016-12-15:
According to the [use page](https://github.com/MatMoul/g810-led/wiki/Use) of your wiki, there is no group for the G-keys.
However, on spec I used the group value `gkeys` analogous to `fkeys` to set the colors on my G910 - which surprisingly worked.
It may be just a typo and `keys` should be `gkeys` but that group should definitely be mentioned, as your software supports it.

MatMoul, 2016-12-15:
sorry wiki need to be updated
gkeys is include in the code

sebschlicht, 2016-12-15:
Yeah, sorry I found it in the `--list-keys` output a few minutes after I've created this issue. My bad. However, at least it's missing in the docs :)

MatMoul, 2016-12-24:
wiki and --help is updated

closed 2016-12-24 by MatMoul