MatMoul/g810-led #29: Project name

MatMoul, 2016-12-20:
I have maked this project for the logitech G810 keyboard but now, it's compatible with G410, G610 and G910...

It becomes interesting to think if the project name should become more generic.

As examples:
logi-led
logi-rgb
other ?

what do you think ?

noisycat, 2016-12-21:
not rgb, since the 610s are technically not rgb. :+1: 

:8ball: ...

I'd put in "logi-light"

But we'll need to make sure and list the keyboards that we can deal with - since I know I stumbled onto the project in a very roundabout way...

MatMoul, 2016-12-21:
@noisycat, You're right, the keyword rgb is not the best ...
I googled "Apex M800" and if this project can be compatible with these keyboards, "logi" is not a good keyword either.

noisycat, 2016-12-21:
I mean, all it would take is for someone to extend the control software scheme to vendors other than 0x046d 

:-D

But actually, let's go with logi-light - I was curious and looked up what else logi could mean - and apparently its old nordish for literally Fire. Fire Light :+1:

MatMoul, 2016-12-26:
Other idea :
kbd-led
lkbd-led (l for logitech)
gkbd-led (g for gaming)

MatMoul, 2016-12-28:
Because this project can not respond to requests for animations, it is necessary to create a daemon that can support this functionality.
For this, I intend to create a new project that work as a daemon and will allow this possibility.
I however need your feedback regarding the name of this next software.

MohamadSaada, 2016-12-30:
Hi, I'm already working on another project and all files will be up in the next week, It has a gui and a daemon for the effects, I have four effects working already, if you want to check the description at <a href="https://github.com/MohamadSaada/Logitech-G-Series-Keyboard-Software">Logitech G Series Keyboard Software</a>, please let me know what you think.

MatMoul, 2017-01-04:
@MohamadSaada Interesting, I wait to see your code ...
Deamon is a very hard work and I don't need it...
If you work on it, it's good for me

Shulyaka, 2017-01-16:
The "LogiLed" seems to be the name that Logitech itself uses in its SDK: http://gaming.logitech.com/sdk/LED_8.87.zip

MatMoul, 2017-01-16:
@Shulyaka Thanks for the info, it's better to not enter in conflict with Logitech...

closed 2017-01-21 by MatMoul

konserw, 2018-01-31:
My proposal is "ledkeyboard" for app and "libledkeyboard" for library - I've used that naming in PR #121 but if it is not acceptable I'll gladly change it to anything else :)

lanodan, 2018-07-30:
What about a name that would also include mouses and potentially other stuff?

I propose button-leds (not sure about where to put the plural btw)

MatMoul, 2018-08-08:
@lanodan Thanks for return, but i encouter now a new problem with mouse...
My code handle only one Logitech device and if you have a G keyboard and a G mouse, g810-led can't manage the two devices...

dioni21, 2020-09-12:
@MatMoul 
> My code handle only one Logitech device and if you have a G keyboard and a G mouse, g810-led can't manage the two devices...

Then you want/need a BIGGER change, where you can specify device paths, vendors, layouts and many other options.

This may not be easy just from a command line and keeping compatibility.  Time to restart from scratch?  This would also make the daemon project easier.