MatMoul/g810-led #30: Wrong executable path in systemd service file

sebschlicht, 2016-12-23:
In the [set profile on boot](https://github.com/MatMoul/g810-led/wiki/Set-profile-on-boot-and-reboot) wiki page when referring to the executable in the `Service` section it is assumed to be located in `/bin`:

    [Service]
    ExecStart=/bin/g810-led -p /etc/g810-led/profile

When the user has followed the installation instructions that are currently present in the wiki, the executable will be located in `/usr/bin`. Thus the section should be changed to:

    [Service]
    ExecStart=/usr/bin/g810-led -p /etc/g810-led/profile

sebschlicht, 2016-12-23:
This applies to both systemd service files present.

MatMoul, 2016-12-24:
Thank for return, I just updated the wiki and the systemd/g810-led.service file.

closed 2016-12-24 by MatMoul