MatMoul/g810-led #35: G610 - NumLock, CapsLock and Scroll Lock leds don't show current state

khnixxo, 2017-01-07:
First of all, great little tool you made here. Thanks!

I don't find anyway to keep the Num Lock, Caps Lock and Scroll Lock leds to show the current state of the keys (on/off)

I have the G610 keyboard.

MatMoul, 2017-01-07:
Normally they function as is.

Can you test by unplugging and plugging your keyboard without using g810-led.
is indicators working?

If not, try to update the firmware with the Windows utility.

If so, it appears that the G610 indicators are not dimmable, it's a possible bug.
To test, you would have to try with the profile group_keys by removing the line indicators.

radare, 2017-01-07:
if you do `-a 000000` the keys lock keys will not be lighted up at all. im with g610 too

radare, 2017-01-07:
you can fix this by running:
```
-k caps ff0000
-k num ff0000
-k scroll ff0000
...
```

i think we should have a different group for those keyus as well as the mm keys and so other indicator ones.

Ideally it would be good to highlight the block keys instead of the indicator ones, and this way be able to use the indicator keys to display network, cpu or disk usage.

khnixxo, 2017-01-08:
Thank you for the quick response.

The issue was the first 2 character (Red on color keyboard?) of the indicators group being 00.
00 turn off the light and 01 to ff control the intensity.

My fix: 
`# group_keys`
`g indicators ff0000`

thanks guys!

closed 2017-01-08 by khnixxo