MatMoul/g810-led #37: g610 orion brown help needed

wermerb, 2017-01-09:
Hi,

first of really thank you for this project. 
Unfortunately it seems I cannot set anything on my keyboard. When I installed it all my keys and numerics went to black and everything else is started to glow white.

These are the commands I tried:
`g610-led -a ffffff`
`g610-led -g keys, ffffff`

MatMoul, 2017-01-09:
Hi,
Firstly, if g610-led -a ffffff not work,
try sudo g610-led -a ffffff
Secondly, the result you describe is the result of de default profile in /etc/g810-led/profile
Try to replace this file with all_red :
sudo cp /etc/g810-led/samples/all_red /etc/g810-led/profile
and reboot

wermerb, 2017-01-09:
Thank you the `sudo` solved my problem.

closed 2017-01-09 by MatMoul