MatMoul/g810-led #38: Script Mode

MatMoul, 2017-01-10:
I think I've found what makes this soft so slow ...
It's to detach and reattach the keyboard to the kernel.

I would need some feedback on the new branch scriptmode.

By using it for simple profile. To see if no keys are missed.

And for those who create effects, by testing the new function before using a script:
g810-led -script-on

For scripts, don't forget that there is the possibility to send several commands with the -pp parameter:
echo -e "k w ff0000\nk a ff0000\nk s ff0000\nk d ff0000\nc" | g810-led -pp

To install :
git clone https://github.com/MatMoul/g810-led.git
cd g810-led
git checkout scriptmode
make
sudo make install

MatMoul, 2017-01-10:
Problem found with scriptmode on, multimedia keys stop working...
you can call g810-led -script-off after your script

MohamadSaada, 2017-01-11:
I solved the problem in my version by making the profile function set all keys in one go rather than calling the detach and attach multiple times, just send a multidimensional array with key names and the colour you want to set, it doesn't have to be for all keys. it can handle from one key to all in one go.
I'm almost done with my software with the daemon and it's looking pretty good, I think It will be ready by the weekend, if you want you can help me there. once it's up I will send you a message.

MohamadSaada, 2017-01-11:
another problem that I faced is that in star effect, if I made too many calls to the attach/detach functions, the kernel will freak at some point and nothing will respond until I hard shutdown and restart the computer.

MatMoul, 2017-01-11:
@MohamadSaada In g810-led you can use -pp to send all keys at once but it's to slow (~150ms)...
For the problem that requires you to restart, look at my Keyboard::sendDataInternal function, now I send an interrupt after the data and I no longer lose control of the keyboard even after a CTRL+C.

MatMoul, 2017-01-13:
hidapi for v2...

closed 2017-01-13 by MatMoul

MohamadSaada, 2017-01-13:
I thought aboud that a while back, because I wanted the one i'm building to work on all platforms including Windows and Mac, the one i'm working on now doesn't work on windows because windows Libusb doesn't support the hotplug functionality.