MatMoul/g810-led #39: General question: How to get G keys and M keys working

CReimer, 2017-01-11:
Is there some Linux tool to make the M and G keys work as they do in Windows or at least something to give them unique events?

MatMoul, 2017-01-12:
At this time, I don't have idea how work G and M keys.

CReimer, 2017-01-12:
There's a kernel module for the G710+ keyboard. https://github.com/Wattos/logitech-g710-linux-driver
But I couldn't get it to work by just switching the ids.

It's the only thing I could find which is even remotely related to the G910.

MatMoul, 2017-02-10:
If possible, report to later...

CReimer, 2017-02-10:
Sorry. I was busy the last couple of weeks. I made some Wireshark dumps of the M-Keys. Please let me know if this helps. You have to filter to 3.2.0. I don't know if it's possible save with this filter already enabled.

[Mkey-Dumps.zip](https://github.com/MatMoul/g810-led/files/768071/Mkey-Dumps.zip)

Filenames are like that. M1-LEDisOff-press.pcapng is a dump of pressing the M1 key while the LED on the key is not on.

The LED switches between the M keys.

The concept behind this is. That you can press MR (MR lights up) press a G-key (MR starts flashing), press any other key to set this G key to the this key and press MR again to save (MR turns off)
M1 to M3 are different profiles.

CReimer, 2017-02-11:
BTW: is there any documentation for these protocols? I'd really like to get the M keys and G keys to work properly. But it's difficult for me to find the right place to start understanding the protocol.

MatMoul, 2017-02-11:
I just looked the dumps but I don't see anything, it's not the same format as my dumps...
Because I don't have a g910, it's difficult for me to understand how G and M keys work...
I don't understant what the problem with M and G keys...
Is the led color but it seem that it is not possible to set the color ?
Is the programmation of G keys but I'm not sure is the best app for this ?
Is MR keys or Mx react on press ?
...

> is there any documentation for these protocols?

Sorry, It's retro engineer...
But you can look here [http://gaming.logitech.com/en-us/developers](http://gaming.logitech.com/en-us/developers)

CReimer, 2017-02-11:
No the M keys don't have a RGB led. It's orange for M1-3 and red for MR
For the key events itself I also think it shouldn't be in g810-led.

The G910 seems to create three devices.

0 Sends key events for G and M keys. But these are not recognized by the Linux kernel.
1 Sends normal keycodes
2 Sends normal keycodes when more than 6 keys are pressed at once.

By default all G keys are mapped to the F keys (G1 -> F1, ...). The Logitech Gaming Software sends some kind of initialization to the keyboard to disable these mapped keycodes, interprets the special key events on the 0 device and translates them for Windows.

frostworx, 2017-03-04:
hi,
today I got my g910. the g810-led project is great, thank you very much for that! (some communication with hyperion would rock :}). I noticed the same wrong behaviour with the G-keys and wonder why this issue was closed.

MatMoul, 2017-03-05:
@frostworx Look the TODO : [https://github.com/MatMoul/g810-led/blob/master/TODO.md](https://github.com/MatMoul/g810-led/blob/master/TODO.md)
I have moved G Keys to "Possible enhancements" because I don't have a G910 and I can not test...
If it's possible to manage G Keys, I need dump or pull request...

frostworx, 2017-03-06:
hi MatMoul, I see. thank you for the quick reply!

frostworx, 2017-03-06:
hi, me again.
no idea if that would help, but I just dumped one "G5" and one "F5" keypress via wireshark
for comparision. looks like some bits are different, so maybe it is a beginning to remap the G keys (f.e. >=F13). the first two lines of both files were written before the actual keypress and therefore are probably useless white noise. if that helps, I'd be glad to capture the other keys as well of course :)

[wireshark_g5_and_f5_keys_g910.zip](https://github.com/MatMoul/g810-led/files/821652/wireshark_g5_and_f5_keys_g910.zip)

MatMoul, 2017-03-06:
I just looked the dump, it can help me but the best is to make a dump when you set your g keys configuration.
At this time, my code is launched one time and don't stay in memory...
I can only set features but not listen for events.

CReimer, 2017-03-06:
I'm not sure if I mentioned this already. But it's acutally the Logitech daemon which handles the key events. If you kill the LGS with the taskmanager the G keys do no longer work.

I suspect it sends some kind of initialization to the keyboard to disable the default behaviour (Gx -> Fx) and that's it. There are special key events for the G keys, but Linux doesn't recognize them on its own. So after we figured out how to disable the default behaviour I can try to build a daemon which redirects the unknown key events to known ones. e.g. F13-F21

frostworx, 2017-03-06:
hi,
unfortunately I do not have any useable windows (vm-)installation, so I can't launch any logitech g-keys configuration (as wine doesn't accept usb devices). by wife has a dual boot laptop, so with some more time I'll take that instead. didn't get very far with both logitech-k750-linux-support and https://github.com/tolga9009/sidewinderd, but I'm not really good in low-level programming so I do not even know if that would be that easy to add 910 support. would love to have it in one application anyway, and as your led control works very nice, I'd prefer that of course :}

frostworx, 2017-03-06:
hi CReimer (OT: afair I know your avatar from some german vdr threads, but I wasn't very active there, so I might be wrong. for whatever reason I even remember a nickname, but do not call it here, it is not "CReimer" :}).
yeah, that was my rough idea as well. detect the real G-keys on low level and redirect the (wrong) F1-F9
to zero or some predefined other keys like F13-F21. if we're there everything else should be already doable within userspace. nice to see communication in here btw, so thank you both! :}

CReimer, 2017-03-06:
You are right. I'm also active with the same avatar as "Copperhead"

frostworx, 2017-03-06:
hehe, I'm really surprised, why I remembered that, as my last comment was in 2012.
looks like you chose both a memorable nick and avatar

CReimer, 2017-03-06:
OK. Here's another dump. This time with Arch Linux and Windows 10 in a VirtualBox. This dump starts with an already running LGS inside the VBox and a device link in Vbox shortly afterwards.

This dump should cover the whole initialization process for the G910

[g910_dump_170306.pcapng.zip](https://github.com/MatMoul/g810-led/files/822576/g910_dump_170306.pcapng.zip)

CReimer, 2017-03-09:
OK. I tried to find my way through these dumps. It's difficult, because I really don't get how these keyboards communicate.
I think I figured out, that all commands to the G910 start with 11:FF:10. What I don't understand is what comes afterwards.

Could you at least try to write a few informations down somewhere to help a newcomer to figure this out? Thanks.

MatMoul, 2017-03-10:
Look at /wireshark_dumps/g810-spectrum/set_keys_colors.pcapng
With this file you can use the next filter in Ŵireshard : usb.dst == "3.2.0" (for your dump, adjust the 3.2.0 value)
My software send data find in URB_CONTROL out, URB setup - Data Fragment
I have found the good data with "Find a packet, HEX value : 00dcff" where 00dcff is the color I have used for my dump

CReimer, 2017-03-11:
OK. I'm getting there.
I found out what to send to enable and disable the leds for the M-Keys

11ff091e01000000000000000000000000000000 #M1
11ff091e02000000000000000000000000000000 #M2
11ff091e03000000000000000000000000000000 #M1 M2
11ff091e04000000000000000000000000000000 #M3
11ff091e05000000000000000000000000000000 #M1 M3
11ff091e06000000000000000000000000000000 #M2 M3
11ff091e07000000000000000000000000000000 #M1 M2 M3
11ff091e00000000000000000000000000000000  #M1-3 off
11ff0a0e01000000000000000000000000000000 #MR on
11ff0a0e00000000000000000000000000000000  #MR off

MatMoul, 2017-03-11:
Very interresting...
I reopen issue with enhancement

MatMoul, 2017-03-11:
Can you try the mkeys branch ?
`git clone https://github.com/MatMoul/g810-led.git`
`cd g810-led`
`git checkout mkeys`
`make`
`sudo make install`
and
`g910-led -mr 1` # or 0
`g910-led -mn 1` # or 0 to 7

CReimer, 2017-03-11:
OK. And here are a few more.

11ff082e01 #Disable mapping of G1-9 to F1-9
11ff082e00 #Enable mapping

Your branch is working btw. But wouldn't it be better to set them as they actually are.
Something like -mn 1 2 Enables M1 and M2

The hex value seems to be a reversed three-digit binary number.

0 --> 000 | M3 off | M2 off | M1 off
1 --> 001 | M3 off | M2 off | M1 on
2 --> 010 | M3 off | M2 on | M1 off
3 --> 011 | M3 off | M2 on | M1 on
4 --> 100 | M3 on | M2 off | M1 off
5 --> 101 | M3 on | M2 off | M1 on
6 --> 110 | M3 on | M2 on | M1 off
7 --> 111 | M3 on | M2 on | M1 on

MatMoul, 2017-03-11:
@CReimer Thank you for your return,
For the -mn, I acknoŵledge but need more code... I will work on it...
Can you confirm that -mnn require a commit (-c) to work ?
And for G to F mapping, I try to include it but need a arg name -? and a return if a commit is required

CReimer, 2017-03-11:
The m key leds don't need a commit and neither does the G->F mapping

And as argument something like:

--gkey-fkey-mapping     Enable or disable the internal mapping of G1-9 to F1-9

MatMoul, 2017-03-11:
--gkey-fkey-mapping can be used if the command is persistant after a shutdown...
If not, is best to use a 2 or 3 char arg for a simple use in profile...
-gkm in relation on the function setGKeysMode

CReimer, 2017-03-11:
Sounds good to me. I wasn't aware that you only use the double hyphen arguments for persistent settings.

MatMoul, 2017-03-11:
Can you try the last pushed version with -mr or -mn, I have removed the commit.
-gkm is implemented with no commit but I not sure if is easy to test the result

CReimer, 2017-03-11:
All three commands work perfectly.

MatMoul, 2017-03-11:
Thank for your work...
If you want figure in contributor, make a PR with your dump in /wireshark_dumps/g910-spark

closed 2017-03-11 by MatMoul

CReimer, 2017-03-12:
Just a little follow up.
I wrote a small python script which checks for the G and M keys and creates proper events via uinput

https://github.com/CReimer/g910-gkey-uinput

It's nice and small but I would prefer to tell the kernel somehow about these keys. So if anyone finds some information about that. Please let me know.

MatMoul, 2017-03-13:
Good work for you g910-gkey-uinput...

[https://github.com/Wattos/logitech-g710-linux-driver](https://github.com/Wattos/logitech-g710-linux-driver) seem the best way to start a driver...

I'm also interested to make a driver but It's more hard to debug...

frostworx, 2017-03-14:
thank you both for your good work! unfortunately I do have too many open projects atm to make something useful with the new build. disabling the G keys works fine. Not sure if I get you right, but as this happens in userspace I wonder how a kernel module should remap the unmapped keys afterwards. wouldn't the kernel code depend on the userspace program then?
haven't actually tested the gkey-uinput python code, but will do so for sure later.

cornernote, 2019-12-19:
@CReimer I'm trying to get M and G keys working on my G815, and I can't seem to find the code to unmap the G keys from the F keys. Are you able to explain what you did while recording to capture that?

CReimer, 2019-12-19:
I'm actually Not Sure how exactly I did it. But my uinput Script relies in the g810-led tool. I found this issue https://github.com/MatMoul/g810-led/issues/39 with all infos on how we figured it out

cornernote, 2019-12-20:
@CReimer #39 is the issue I contacted you from :)

I have tried sending everything the windows software sends on startup to the USB, using this test script:

https://gist.github.com/cornernote/805de3ae9d50922e1769ef78296da60d

Nothing seems to enable the G-Keys :(

I basically got the wireshark dump, extracted everything from the startup before the colour settings, then ran that in a python script.

Is that how you found it?

CReimer, 2019-12-26:
> @CReimer #39 is the issue I contacted you from :)

Oops 

> 
> Is that how you found it?

I hacked it into @MatMoul's g810-led. But yes. That's more or less how I figured it out.

The first thing you need to figure out is how to disable the default mapping. I don't expect it to be much different (if at all) from the G910 keyboard.

You need g810-led up and running with the -gkm 1 parameter. This disables the Gx --> Fx mapping. From now on the G-keys send their own keycodes. To make these known to the system, you need the g910-gkey-uinput Python script.

As far as I remember I often watched the wrong device with Wireshark. To work around N-Key-Rollover issues there are two keyboard devices. And the special keys M and G are only available on the second device

cornernote, 2020-01-07:
@CReimer I found the code to enable the G-Keys!

I have documented everything here:
https://gist.github.com/cornernote/82972cc307985df29f3ddca214a4dc39

It's also worth linking to the issue for the G815 keyboard - https://github.com/MatMoul/g810-led/issues/188

I plan on adding G815 support to sidewinderd for full G-Key support - https://github.com/tolga9009/sidewinderd/issues/40

commit 019c848d (2017-03-11): Add wireshark dump which helped fix #39
Frame 3955: Disables the default mapping of the G keys. I figured 00 out by myself to reenable the default mapping
Frame 4397: Enables M1. Everything else was trial and error. More info in #39
Frame 4401: Disables MR. Figured 01 out via trial and error.
commit dcdda2dc (2017-03-11): Merge pull request #51 from CReimer/master

Add wireshark dump which helped fix #39