MatMoul/g810-led #40: G403 Wireless mouse support

Shulyaka, 2017-01-12:
Hi!

I now own a G403 wireless mouse and would like to control its LEDs. This project is the closest software I could find to this goal, but it only support keyboards. Hopefully, the protocol is close enough.

So this is more of a question of how to add support for a new device to g810-led. What is the process? Are there any specs available from Logitech? Or should I find a Windows box, install original Logitech software, then start sniffing USB or so, trying to guess what the protocol is?

MatMoul, 2017-01-12:
Hi,
For the keyboard, I used Windows with le Gaming software and wireshark...
The best way to sniffing is to use a second mouse to control wireshark to minimise data sent to the G403.
For the first dump, I haved set all keys to a defined color (ex green : 00ff00) and after I searched for the color in the dump.
With this you will find the command to send, but for the keyboard, I need to send a second command to validate change (commit function).
I have found this command on the next send line in dump.
For testing it's best to use my python version [https://github.com/MatMoul/g810-led-python](https://github.com/MatMoul/g810-led-python). The code is more fast and easy to edit for retroengeneering.
Good luck

Shulyaka, 2017-01-16:
Hi,

Thanks for the advice, I have finally managed to get a first working code that switches the leds on my mouse.

Now this is the question to you as original author: do you want me to integrate my changes to your tool or maintain a fork? The reason I ask is that it would be a significant change and the code would not be as simple as it is now because:
* I have found out that g403 uses a different protocol than g810 and g910 keyboards. It also doesn't need a commit.
* Mice and keyboards are different, and the number of supported functions also differ. For instance, we don't have horizontal or vertical wave effects with the mouse and only have two leds. At the same time I have plans to read battery level because it's a wireless device.
* A user may have both supported keyboard and mouse, therefore the current device selection mechanism based on vendor_id:product_id is not enough.

Anyway, I think I will make a PR when ready and then you'll decide. It will take some time for me.

closed 2017-01-16 by Shulyaka

MatMoul, 2017-01-16:
@Shulyaka Thank you for your return, it is very interesting ...
Before make a PR, look on v2 branch, I have refactored the code and based it on hidapi lib.
v2 is a research branch, don't make PR... (give me one or two week).
If all working good, we could start a new common project that support new Logitech hardware by becoming the next gnome15 :-p
Just need to found a good project name...

MatMoul, 2017-01-22:
@Shulyaka I have refactored the code and now, hidapi lib is used as default...
now, you can try to work on a PR...

spiritg9, 2017-10-16:
Hello, 

as a owner of both, G810 and g403 I would like to contribute here. Wanted to check if there is any code for g403 I could start with?

Also, sry for off topic but don't want to open new issue. Have you thought about integrating your software with some other applications like telegram or alsa/pulseaudio. It would be really cool for some keys to change color according to incoming messages or volume.

Shulyaka, 2017-10-17:
Hi,

I have a proof-of concept code on my desktop (not pushed to github yet). Unfortunately due to family issues I had no time to work on it, but now that I realize I'm not the only one wanting the G403 support, I'll try to find the time this week to rebase my code into current version of g810-led. Sorry for the delay.

spiritg9, 2017-10-17:
@Shulyaka 
Great, thx!

If there is anyway I could help, pls let me know.

anchepiece, 2017-10-24:
I would love to see mouse support as well.  I'll try to help with testing as well as running captures if needed.

Shulyaka, 2017-10-25:
Please see #106