MatMoul/g810-led #46: g610 brown help required

babhijit, 2017-02-04:
When I tried to set up fx effect using the command:
        g610-led -fx hwave keys ff

The entire keyboard became dark. Thanks in advance for help :)

MatMoul, 2017-02-04:
Thank for your return...

At this time, -fx is confirmed only on G810 because I don't have other model for test it.

Have you tested with other value than ff?
And for other effects? Do they work?

MohamadSaada, 2017-02-07:
Hi MatMoul, sorry for hijacking the thread, but there is no other way of contacting you, I've uploaded my code for the GUI under [LogiGSK](https://github.com/MohamadSaada/LogiGSK), if you can check whether it works on your device or not, I would really appreciate it.

MatMoul, 2017-02-08:
@MohamadSaada I will try this week-end if I found time...

CReimer, 2017-03-11:
Just to let you know.
All effects work on my g910 just fine.

MatMoul, 2017-03-11:
@CReimer Thanks for your return

MatMoul, 2017-03-13:
@babhijit I close this issue because I don't have return... but you can reply me here if you encouter problem

closed 2017-03-13 by MatMoul

jpl81, 2017-06-29:
I also have a G610. The hwave, vwave and cwave effects don't work on my g610 irrespective of the value given. The cycle effect works but the value of speed does not seem to make any difference. Other effects (color, breathing) work as they should.

MatMoul, 2017-06-29:
@jpl81 Thank for your return, but taht is an other issue...
I'm missing time thes days but open a new issue for this case.