MatMoul/g810-led #47: Question: support select keyboard

janoshh, 2017-02-08:
Hi MatMoul! I want to connect 2 keyboards and select the keyboard I want to write the effect/keycolor to. Is this possible or can u give me a hint how to do this?

MatMoul, 2017-02-08:
At this time this project set only the first keyboard found...
If you have two different model (ex g810 + g910), it's quite simple to change the keyboard detection code according to the binary name.
But for same model, it's more difficult... Need argument to provide serial or index to select the keyboard...

janoshh, 2017-02-08:
Thanks for the reply! I'm gonna take a look at de hidapi, maybe i can figure it out.. It needs to be more speed between shifting the colors, any idea why the error apears with hidapi?

MatMoul, 2017-02-08:
I did not understand everything...
Normally hidapi is very fast ...
In what condition does this happen?

janoshh, 2017-02-09:
Yes hidapi is very fast, but when i tried it, it gave me an error: writing on hidraw is not allowed.. it appeared when i connected two keyboards.. The effects work much faster then setting the keys individually, how come?

MatMoul, 2017-02-09:
Very strange that "writing on hidraw is not allowed" appear when you connect a second keyboard but I never tested.
For effects, I use native effect in the keyboard... this is why it is more speed to set...
But if you want to make custom effect, don't forget to use the pipe method (-pp)
You can look at the k2000 sample with speed 0 (sample_effects/bash/k2000 0)

janoshh, 2017-02-10:
allright i understand! I have another question, what data does the keyboard expect? I tried to figure out the wireshark dumps but no succes..

MatMoul, 2017-02-10:
I just tested with two keyboard and I don't receive "writing on hidraw is not allowed". What's your linux distro?
For dump, in wireshark_dumps/g810-spectrum/set_keys_colors.pacapng, I filter result with usb.dst == "3.2.0" where 3.2.0 is my usb port...
And I found data in URB sertup - Data Fragment

janoshh, 2017-02-10:
I'm running it on raspbian (Raspberry Pi). I'm not always getting the error, somtimes it's not setting every key when i set all keys.. And thx, I'll check the dump again!

MatMoul, 2017-02-10:
Interesting, I have never tested on rp...
I have few rp on my job, I would test when I had the opportunity...

MatMoul, 2017-02-10:
Because RP has a slow CPU, one posssible bug is the usleep at line 509 of keyboard.cpp...
Try to increase value...

janoshh, 2017-02-14:
Hi MatMoul, sorry to bother you again! I'm a noob in C++, but I can understand the layout etc from C# etc.. I see in Keyboard.cpp that u are getting the keyboard ID (is this unique?), how can I acces that ID and do a check in the main.cpp?

MatMoul, 2017-02-16:
I'm not sure what you call KeyboardID but if it's the productID, I'm using lsusb.
I have put the value back from my G810 keyboard and for other models, I thank the contributors.

closed 2017-04-29 by MatMoul