MatMoul/g810-led #49: Keys with no color after script / profile

asathoor, 2017-02-28:
When I run a `g810-led -pp < profilefile` some random keys are not colored. I have to repeat once, sometimes twice before all keys are as I want them to. On the fly it's not a big issue, but when I try to color the keys via `rc.local` or similar it's an issue.

MatMoul, 2017-03-01:
Seem the same problem than the second part of #45

MatMoul, 2017-03-03:
Do you have the same result with "g810-led -p profilefile" ?

asathoor, 2017-03-10:
Yes sometimes. However that was yesterday. I'll test out the recent version of the code now.

asathoor, 2017-03-10:
Yes. The issue is the same: Tried some `g810-led -p all_blue` etc from the sample files. The problem is the same. Some random keys are not colored. Right now my keybard is a mixture of red and green keys afer a `... all_green` followed by a `all_red`.

* OS: Linux 4.8.0-39-generic #42-Ubuntu SMP Mon Feb 20 11:47:27 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux

MatMoul, 2017-03-11:
Sorry, I can not reproduce but it's a bug...
Try the solution of @CReimer on issue #45

MatMoul, 2017-03-11:
@asathoor Seem to be fixxed in the last version...
Can you confirm ?

MatMoul, 2017-03-13:
I have two confirmations, I close the bug.
If bug persist, I will reopen it

closed 2017-03-13 by MatMoul

asathoor, 2017-03-14:
Perhaps it's a "freak of nature". But on my system it's still a problem.

asathoor, 2017-03-14:
After an update a minute ago everything works fine. I tried to run the script from rc.local - but if the Ubuntu GUI for upstart programs is used everything works fine. So I guess that's the proper way to run g810-led on Ubuntu.

asathoor, 2017-03-14:
Now everything is ok. I changed the script from rc.local to the standard Ubuntu add programs on upstart. Now everything works as a charm.

MatMoul, 2017-03-16:
Thank's for your return
It's cool to hear that..