MatMoul/g810-led #50: Key Press / Fade

fireAnon, 2017-03-05:
Hello! Just wanted to say thank you so much for uploading this, very simple easy to use, I love it. Was playing around though and found im having a hard time getting a key press then fade back to original color effect. Logitec offers: _**Key Press** — The keys change color when you press them. Use the color selection wheel to choose a preset color or customize your own. You can also set a background color for the entire keyboard. Use the slider to set the rate at which the color fades back to the background color._

I was curious If you had any Ideas on how to make this work, also I found I cannot find a way to set a color to my numberpad_enter key, I've tried _g810-led -k num_enter 00ffff_ and _g810-led -k enter 00ffff_ but neither seem to work for me.

MatMoul, 2017-03-05:
Try numenter : g810-led -k numenter 00ffff
You can find all keys with g810-led --help-keys

fireAnon, 2017-03-05:
that worked, yea I see it now sorry I missed that, as for the key press, is that not possible right now? Thank you for the quick response @MatMoul

MatMoul, 2017-03-05:
Indeed for the "KeyPress" it is not possible for the moment ...
The future evolutions are described in the TODO.md file
[https://github.com/MatMoul/g810-led/blob/master/TODO.md](https://github.com/MatMoul/g810-led/blob/master/TODO.md)

closed 2017-03-05 by MatMoul