MatMoul/g810-led #56: Question: Add a new repo for custom effects

wextia, 2017-03-23:
I was taking a look at [TODO.md](https://github.com/MatMoul/g810-led/blob/master/TODO.md) for ideas to contribute to, and i stumbled upon this one. Can anyone explain a bit how can i start implementing this? I like the idea so much. :smile:

MatMoul, 2017-03-23:
Before starting a new repo for custom effects, I think we should solve a few points...
* Provide a solution for multiple keyboards support
* Provide a idea for keyboard layout support
* Try to find a more generic name for the project
* Add a deamon mode for custom effects

MatMoul, 2017-03-23:
@wextia Don't miss to add your PR contribution in CONTRIBUTORS.md

wextia, 2017-03-24:
Huh, ok. Could you explain those points a bit further? I'd like to help, but since this points are kinda vague i don't know from where to start! :cry: 

Also, thank you very much for your work. I am using a g810 and it has been so much fun twiddling with it thanks to you and the contributors :smile: .

MatMoul, 2017-03-24:
I'm in full reflexion for multiple keyboard support and keyboard layout support.

Firstly, we need common args for all effects to support the three next possibilities :
* add optional arg for keyboard index provided by --list-keyboards
* add optional arg for keyboard serial number
* add optional arg for keyboard layout
For these three possibilities, we don't need a functional solution, but a concept.

Secondly, we need a deamon mode to support start, stop and change command for effects because custom effects is not a simple command but a script who reside in memory.

Thirdly, it would be good to find a more generic name because I have maked this soft for g810 but now it can handle many models and it's not clear for other model

MatMoul, 2017-04-03:
Sorry for the late reply...

Firstly, your code for the daemon mode is a good base.

Secondly, I have tested your PR  #59 and my indicator flashes when I use 2 effect. (Need more tests before provide a multiple effects solution.)

Thirdly, custom effects need to be independent script that can be scripted in bash, python, ...

MatMoul, 2017-04-06:
One of my idea for custom effect is to make scenary file that support all feature in profile file.
And add possibility to call profile file, use timeout, loop, ...

MatMoul, 2017-05-01:
Now g810-led support keyboard selection by serial number, vendor id and product id.
The new code can be support easily to add a parameter to select keyboard by index or to handle keyboard layout.
The three first point are ready for the next step.

joaogma, 2017-05-06:
Very off-topic, but do you have plans to support the new G Pro keyboard?

MatMoul, 2017-05-06:
@joaogma Your right, it's very off-topic...
Please open a new issue

MatMoul, 2017-05-12:
I have open a new repo [https://github.com/MatMoul/g810-led-resources](https://github.com/MatMoul/g810-led-resources) for profiles and dumps...
Thereafter, custom effects will be available in this depo

closed 2017-05-12 by MatMoul