MatMoul/g810-led #58: Effects on G410

Landrovan, 2017-03-24:
Here is my test of the effects on the G410. I use the sample provided in /etc/g810-led/samples/.

all_blue_fxl_breathing_red: Does the fx only on the logo. So it don't work on the G410 (no logo to light).
fx_breathing_red: Work perfectly
fx_color_green: All keys became green. And then nothing change. I don't understand this fx. What is it supposed to do?
fx_cwave: Works perfectly. The wave start around the 'uijk' key.
fx_cycle: Works perfectly.
fx_hwave: Works perfectly.
fx_vwave: Works perfectly.

If you want video to see real result, just ask.

Note: In the --help-effects a little information on each effects would be good.

MatMoul, 2017-03-24:
Thank's for your return.

For all_blue_fxl_breathing_red, your right, but samples effects are for all models.
I need to remove logo to g410-led --help-effects

for fx_color_green or -fx color, it's a native effect that set all keys to a color.
You can test the differance:
`time g810-led -fx color all 00ff00`
`g810-led -k g ff0000`
and
`time g810-led -a 00ff00`
`g810-led -k g ff0000`
Result and speed are different.

MatMoul, 2017-03-24:
--help-effects corrected for g410-led

closed 2017-03-24 by MatMoul