MatMoul/g810-led #60: G610

antigol, 2017-04-02:
Hello,

I have a G610 and the native effects does not work properly.
For instance the cycle effect turn the light on following a gradient but there is not time evolution.
How can I contribute to fix this bug ?

Also on windows it is possible to have an effect where the keys light turn on for a small time after pressing them. Is it possible to implement this effect ? 

If I well understand, these native effects are implemented in the hardware and we can run them by sending a simple message via USB. Is this effect "light on when pressed" also native ?

Also is there any memory into the keyboard to remember the configuration ?

MatMoul, 2017-04-03:
Hello and thank you for your return.

Firstly, the first job for fix this bug is to make a wireshark dump when you set your effect with the logitech software.

Secondly, for testing if effect is native, set your effect with the logitech software and exit it.
If the effect work without the logitech software, it's a native effect.

MatMoul, 2017-04-29:
I close this issue because I have no returns

closed 2017-04-29 by MatMoul