MatMoul/g810-led #62: Expose underlying interfaces as a shared library

pearsonk, 2017-04-05:
It would be useful for the interfaces exposed by Keyboard.cpp to be exposed as a shared object library so that other applications could link against the underlying implementations without the need to create the driver process (g*10-led). The segmentation between the driving application (main.cpp/helpers) and underlying implementation seems fairly conducive to creating a library.

MatMoul, 2017-04-06:
I have started this code without library because it's more easy fore installation but it's a good idea to make it

pearsonk, 2017-04-07:
At a decision point in the implementation. I have the library building and deployable, so what now remains is the default behavior of the g810 application.

Would it be preferable for the g810-led application to be statically linked (current behavior), or dynamically linked using the library?

Trade-offs here are that use of dynamically linked binary generation will require an extra step when testing (building but not installing), as an expense of having universal test coverage of the library interface. Currently I am putting the libg810-led.so in a lib directory of the build environment, so running the application prior to install will require modification of the LD_LIBRARY_PATH to include this directory. example:
`  # sudo LD_LIBRARY_PATH=./lib ./bin/g810-led <COMMAND>`
On install, this would be transparent to the user and they could run /usr/bin/g810-led without this additional overhead. 

Thoughts?

MatMoul, 2017-04-09:
Thanks for your work.
At this time, it's seem better that g810-led stay with the current behavior, that's more easy to debug and deploy...
But providing a way to build the lib for the futur is very cool (make lib)

closed 2017-04-14 by MatMoul