MatMoul/g810-led #64: Reduce needed sleep time for G910

CReimer, 2017-04-05:
Follow-up issue for #63 

Things I didn't test:
There might be a connection between what's been sent to the keyboard:
- A commit might take the full 1.1ms or depending on how much is cached to commit.
- Pure RGB commands might not need this sleep at all.

But for this I need your help. Where's the best place to try different usleep times for different commands. Especially the difference between the commit command an non commit commands?

Are you interested in my python based test script?

CReimer, 2017-04-06:
OK, after a closer look. I think I need to rework the sendDataInternal function a bit. At least for testing purposes. I need a second parameter for the sleep time. If I come up with a good solution I'll send a new pull request.

CReimer, 2017-04-06:
#63 seems to be the only solution. 1.1ms is the maximum speed possible.
I tried to increase the time for a commit and decrease the time for a color command: Immediate firmware crash.

closed 2017-04-06 by CReimer

MatMoul, 2017-04-06:
Thank for your test,
sendDataInternal is the more difficult part of the code...