MatMoul/g810-led #67: Asterisk symbol not working on Nordic keyboards

zone31, 2017-04-07:
When using single key operations (-k) on my g810 i can set all other keys, except for the asterisk(*) key on my keyboard. When setting the locale to English, the key becomes backslash(\). When using the key group, the color changes correctly. I have tried g810-led -k backslash ff00ff , which did not change it.
I am using the (0x46d, 0xc337) model g810.
I have added images of my keyboard in english locale, and Nordic(Danish) locale
[image]

[image]

MatMoul, 2017-04-07:
Try with intl_backslash
or test all possibility in g810-led --help-keys (in Group keys)

zone31, 2017-04-07:
got it to work, it is called "dollar" in the project. 
Thanks for the help :)

closed 2017-04-09 by MatMoul