MatMoul/g810-led #69: G213 Support

pearsonk, 2017-04-17:
The Logitech G213 is a new LED keyboard that supports many of the core features, with the exception of it having 5 LED regions as opposed to the per-key-LED structure of the other currently-supported keyboards.

Some preliminary work necessary to investigate feasibility: 
https://github.com/pearsonk/g810-led/tree/G213-develop
Only took a swing at setAllKeys for now, but split across 5 calls to a new setRegion function.

MatMoul, 2017-04-18:
@pearsonk I see that work hard...
It seem you are almost ready with the code...

pearsonk, 2017-04-19:
Yes, turned out to not be terribly difficult to integrate. It's very similar to the other keyboards. I'll have a rebased PR soon.

MatMoul, 2017-04-21:
Yeah, i enjoy, very good job...

closed 2017-04-21 by MatMoul

MatMoul, 2017-04-29:
@pearsonk I come back to you for the g213,
If I understand correctly, this one does not support arguments -g -k -an -gn -kn -c
Can we delete them from the help for this keyboard?

pearsonk, 2017-04-29:
Yes, you are correct. I was thinking of a way to better handle this that didn't result in an incomprehensible mess of if statements, which will get very complicated as more devices are supported. Perhaps a capabilities bitmap that could be set for each keyboard model. Then capabilities could be checked instead of specific device models, such as:
`if (kbd.getCapabilities &= LedKeyboard::Capabilities::PER_KEY_RGB) doSomething())`

MatMoul, 2017-04-29:
@pearsonk I agree that the current method to filter features in help are difficult to maintain.
Your solution seem to be good.