MatMoul/g810-led #72: Question: Timeline for libusb-1 deprecation

pearsonk, 2017-04-26:
I'm working on some code to address issue #47 and support for specifying device by product ID as well as serial (and fix a current wchar_t logic error). This will allow running commands and automate testing of multiple keyboards without having to plug/unplug devices. 

I have this working with the hidapi-hidraw portion of the code and was wondering what the timeline is for dropping the libusb-1 portion. If it is to be dropped soon, I'd rather not invest time in writing/testing that particular interface.

MatMoul, 2017-04-26:
At this time, I don't have many return for hidapi (seem to be a good signe)...
But the primary raison to maintain it is that hidapi don't work on CentoOS (write is not allowed on usb with hidapi but is allowed with libusb).
If someone finds a solution, for me, libusb support can be removed.
If your solution skip the param without error with libusb, it's OK for me.

MatMoul, 2017-04-26:
Just a simple remark, it would be nice we can select keyboard by serial number as well as index provides by --list-keyboards.
One of my idea is that a binary with a more generic name that g180-led work with all supported keyboards and all other hardlink with a defined name, list only keyboards of the binary name model.

pearsonk, 2017-04-27:
I have this mostly implemented in https://github.com/pearsonk/g810-led/tree/SpecDevice
With this I no longer need to even use hard-linked program name except for the help output. I implemented the libusb portion as well, and fixed a couple segfault issues in the libusb implementation.

My next step is to modify listKeyboards to provide a DeviceInfo object which contains the vendor ID, product ID, serial number, and LEDKeyboard::KeyboardModel, plus modify listKeyboards to provide a vector<DeviceInfo> that can be parsed by main.cpp but would also be useful to implementers of the library which really shouldn't be touching stdout. This should make it fairly simple to query the list and allow selection from the vector without having to re-scan the USB devices to find a match.

I had to make a fairly drastic change to main.cpp command line argument parsing to be more adaptable to specifying device constraints. Please let me know if the approach isn't ideal. I kept it as a separate commit so that it can easily be extracted.

I will close this ticket out and issue a PR once I can finish the last bit of development and testing.

closed 2017-04-27 by pearsonk

janoshh, 2017-04-28:
@pearsonk Awesome! So now I can pass the serial number of the keyboard when setting for example key "a" to blue? What is the exact syntax to do this? I want to edit the code so that I can select it from within the code (i'm not using the command line to set the keyboards).