MatMoul/g810-led #75: Logitech G Pro keyboard support

joaogma, 2017-05-06:
The title is self-explanatory.

Logitech released this keyboard in March, and it is pretty much a TKL G810.

Any plans to update the g810-led with G Pro support?

Thanks for your wonderful, groundbreaking work with the supported keyboards, I never thought I would find something like this!

MatMoul, 2017-05-06:
Thank you for your return...
I don't have this model but it is very similar from supported models.
For the other models than the g810, it's other contributors that has provided the support.
If you have already buy your keyboard, you can become a contributor to help adding the support.

If your interrested, return me the result of the next command :
lsusb | grep 046d

The next step will be to test the supported protocols by modifying Keyboard.h

joaogma, 2017-05-06:
@MatMoul Unfortunately I don't have one...

MatMoul, 2017-05-06:
So we'll have to wait for someone to return me the necessary information :-(

MatMoul, 2017-05-12:
Pull Request merged and new version with gpro support published !

closed 2017-05-12 by MatMoul