MatMoul/g810-led #77: Question: KeyboardFeatures and devices

pearsonk, 2017-05-11:
I was looking at the KeyboardFeatures implemented in help.cpp, some of the feature assignments got me thinking. Is the current feature assignment (indicators, multimedia, numpad, etc) intended to indicate that the group is physically present or that it has software-programmable LEDs.

To give an example, the G910 has multimedia keys and they have LEDs, but they are not programmable and are always the default blue.

MatMoul, 2017-05-12:
At this time, this is intended to software-programmable LEDs

MatMoul, 2017-05-12:
@pearsonk Following your idea, the implementation of features flags in help.cpp is a first shot and a good solution.
Now, It would be nice to move this solution to Keyboard.cpp.

pearsonk, 2017-05-13:
Yes, I already started a little migration but haven't had time to focus on it. In addition to migrating much of the present bitmask, I'm incorporating the definitions alongside SupportedKeyboards but expanding the size to uint32_t to give some room for growth. Currently there are 14 of 16 bits spoken for in a uint16_t.
There are also a few capabilities errors that I have encountered in the keyboards I have available for testing {G213, G410, G510, G510S, G810, G910}. I do not have a G610 or GPro to validate those so I will go based on the defined code.
More to follow, perhaps late next week.

closed 2017-05-13 by pearsonk

MatMoul, 2017-05-13:
I look forward to your contributions

MatMoul, 2017-05-15:
It would be nice to have a similar solution for keyboard protocols.
The help solution is a good test to see what we need.
To think about it ...