MatMoul/g810-led #78: Daemon with scenarios

MatMoul, 2017-05-13:
I just pushed a new branch named daemon that support scenario file.
Scenario file can contain all commands available in profile file and also :
* w time : wait time in millisecond
* p filename : load a profile

To launch it, call :
g810-led --daemon scenarioFileName
To kill it, call :
kill -2 pid (pid is returned when you launch daemon)

Returns is welcome

MatMoul, 2017-05-15:
Now I have tested scenario0 but it don't work very well when you use your keyboard.
I have added a k2000 scenario who work well.

Clarification : This branch is not intended to be merged as is. But for testing this solution.

Small note for scenarioFileName: Because it's a daemon, absolute paths must be used.

MatMoul, 2017-05-20:
Now scenario0 work well.
Need tester for returns.

PFCKrutonium, 2017-05-21:
Okay, so I built this branch, and ran
`./bin/g810-led --daemon ~/sample_scenarios/k2000 ` - Which seems to have done nothing?

MatMoul, 2017-05-21:
Thank you for your return.

At this time only the p and w commands are available.

A first possibility is to use a second script to generate the scenario in a cron or other trigger.
When daemon is launched, you can update the scenario file without restarting the daemon.
As exemplle, you can test by updating waittime k2000

I also provided a unit that copy the scenario file in /tmp/g810-led-scenario
When the unit is started just update this file to change the effect.

Now I think about how I can handle more possibilities (calling bash or python script) but the dilemma is the security.

closed 2018-09-13 by MatMoul