MatMoul/g810-led #80: gpro loses key settings

lleukkun, 2017-05-16:
My gpro is occasionally switching key colors or turning them off completely. Sometimes just one, but just had most of the regular keys go dark. Re-running gpro-led fixes this, but perhaps there's some power management thing kicking in? I will try to investigate this more.

MatMoul, 2017-05-18:
It's possible that this bug is the same that #81

MatMoul, 2017-05-19:
@lleukkun I just pushed a new branch named "bug_sometime_wrong_key_color" with an updated sendDataInternal function.
Try with this branch...

MatMoul, 2017-05-20:
Branch "bug_sometime_wrong_key_color" is now merged to master.
Just install the last version

lleukkun, 2017-05-23:
I can't trigger this anymore, running currently with the latest. If it comes back I'll reopen.

lleukkun, 2017-05-23:
And it happened again about 5 minutes after closing this ticket, running version 0.2.5 (commit 99447196561c5e083e7bd1f2fda69d8cebdab4fd). Reopening.

FruityFaiz, 2017-05-30:
For now I'd suggest you set your colours in Windows, if I remember right, the GPRO has on board memory so it should save it.

MatMoul, 2017-05-30:
@FruityFaiz Thanks for your return.

@lleukkun It can be good to test setting keys with windows software and see if your keyboard lost color in linux after a certain time.

lleukkun, 2017-05-31:
gpro may have memory but it reverts to the same default effect after reboot so doing things in windows doesn't really make any difference. We should probably just sniff what the windows software is doing over usb and see if there's something that stands out.

MatMoul, 2017-07-14:
@lleukkun What the state of this issue ?
No news, good news ?

MatMoul, 2017-08-30:
@lleukkun 2 month without reply, reopen issue if needed

closed 2017-08-30 by MatMoul