MatMoul/g810-led #81: Left Alt is Green?

PFCKrutonium, 2017-05-17:
```
g logo 000096       # Blue
g indicators ffffff # White
g multimedia 009600 # Green
g fkeys ff7700      # Orange
g modifiers ff7700  # Orange
g arrows 0000ff     # Red
g numeric 00ff00    # Green
g functions ffffff  # White
g keys ff00ff       # Purple
g gkeys ffffff      # White
k tilde ff0000      # Red
k W 0000ff          # Blue {
k A 0000ff
k S 0000ff
k D 0000ff          # }
k 1 ff0000          # Red {
k 2 ff0000
k 3 ff0000
k 4 ff0000
k 5 ff0000
k 6 ff0000
k 7 ff0000
k 8 ff0000
k 9 ff0000
k 0 ff0000          # }
c                   # Commit changes
```
This is the profile I am using right now. It all works (mostly) correctly - somtimes some keys don't change (my F Keys didn't turn on last reboot), but my left alt is consistently green. Any ideas what's up, or is this a bug in the software?

PFCKrutonium, 2017-05-17:
Temp Fix:
`k alt_left ff7700`

MatMoul, 2017-05-17:
Very strange, do you use the last version?

PFCKrutonium, 2017-05-18:
Yep, in fact, I am using the Arch (AUR) g810-led-git package - which is build from the repo.

MatMoul, 2017-05-18:
At this time, I can not reproduce, need more investigation...
I try to test your profile in the next few days

MatMoul, 2017-05-18:
I also encounter some problem with your profile.
Now, I only have to search :(

MatMoul, 2017-05-18:
After small tests, I suggest to add a commit (c) after line 10 (g gkeys ffffff      # White).
Set all group, commit, set keys and commit.
If it work, I need to update code for handle this case.

PFCKrutonium, 2017-05-19:
Okay, giving this a shot.

PFCKrutonium, 2017-05-19:
My F keys still do not light up by default, and now my menu key (the key to the right of the right Windows key) has not lit up either.

PFCKrutonium, 2017-05-19:
Current Profile as of now:
```
# Sample profile by groups keys

g logo 000096       # Blue
g indicators ffffff # White
g multimedia 009600 # Green
g fkeys ff7700      # Orange
g modifiers ff7700  # Orange
g arrows 0000ff     # Red
g numeric 00ff00    # Green
g functions ffffff  # White
g keys ff00ff       # Purple
g gkeys ffffff      # White
c                   # Commit
k tilde ff0000      # Red
k W 0000ff          # Blue {
k A 0000ff
k S 0000ff
k D 0000ff          # }
k 1 ff0000          # Red {
k 2 ff0000
k 3 ff0000
k 4 ff0000
k 5 ff0000
k 6 ff0000
k 7 ff0000
k 8 ff0000
k 9 ff0000
k 0 ff0000          # }
# k alt_left ff7700   # BUG!!! Has to be manually set!
c                   # Commit changes
```

MatMoul, 2017-05-19:
No luck...
You can try temporarily with libusb. (reinstall the package and edit the PKGBUIL)
Concerning your profile, it is correct for me.
You can also remove the intermediate commit (c) that I suggested, it is useless.
I investigate...

MatMoul, 2017-05-19:
I have pushed a new branch named "bug_sometime_wrong_key_color" that contain a fix.
Can you test ?

PFCKrutonium, 2017-05-20:
Will give that a shot

PFCKrutonium, 2017-05-20:
Okay, so using that branch without rebooting (Installing some stuff that takes a while, will update afterwards) Appears to have worked, except that my fkeys remain turned off for some reason.

MatMoul, 2017-05-20:
fkeys work well for me...
Try to unplug and replug your keyboard.

PFCKrutonium, 2017-05-20:
Nope, It visibly reset but didn't correct the F keys.

MatMoul, 2017-05-20:
Just to be sure, when you have installed the branch "bug_sometime_wrong_key_color" you have only replaced the line 19 in PKGBUILD with :
source=("git://github.com/MatMoul/${_appname}.git#branch=bug_sometime_wrong_key_color")

PFCKrutonium, 2017-05-20:
I didn't use the pkgbuild, I did it manually
```
[krutonium@krutonium-desktop g810-led]$ git status
On branch bug_sometime_wrong_key_color
Your branch is ahead of 'origin/bug_sometime_wrong_key_color' by 1 commit.
  (use "git push" to publish your local commits)
nothing to commit, working tree clean
[krutonium@krutonium-desktop g810-led]$
```
I ran `make bin` and copied the resulting executable to `/usr/bin/g810-led`

MatMoul, 2017-05-20:
And what happens when you make a
g810-led -k f1 ff0000

PFCKrutonium, 2017-05-20:
That also worked.

MatMoul, 2017-05-20:
Your build method is ok
But is it normal that your branch is ahead of ... by 1 commit ?
And what appens when you make a 
g810-led g fkeys ff7700

PFCKrutonium, 2017-05-20:
Honestly I don't know why it says that, I did a `git reset --hard` and it still gives me that message.

```
[krutonium@krutonium-desktop g810-led]$ g810-led g fkeys ff7700
```
Outputs usage/help.

MatMoul, 2017-05-20:
sorry...
g810-led -g fkeys ff7700

PFCKrutonium, 2017-05-20:
Yep, that worked correctly.

MatMoul, 2017-05-20:
Very strange.... the -g fkeys is the same function that is used in profile file.
No idea for now.
I will make a new release with the last changes and we will investigate from this new base

PFCKrutonium, 2017-05-20:
Okay. Should I make a new issue for the F keys?

MatMoul, 2017-05-20:
Yes it will be good

PFCKrutonium, 2017-05-20:
Will do.

closed 2017-05-20 by PFCKrutonium

MatMoul, 2017-05-20:
Thank for your work, you can make a PR with your profile in sample...
It's a good profile for debug the code

PFCKrutonium, 2017-05-20:
Sure.