MatMoul/g810-led #82: Procedural vs object hierarchical organization

pearsonk, 2017-05-17:
There are talks to add support for non-Logitech keyboards, and as the number of supported systems in the current procedural implementation is causing a lot of if/switch parsing based on the KeyboardModel which will only get more complex as the list grows and they types of keyboards and protocols grows.

An alternative approach would be to disaggregate Keyboard.cpp implementations into separate classes, with protocol implementation in an intermediary, and model-specific implementation in a model-specific class.

I believe most of this work could be done without much disruption to the library API. It would be quite an overhaul, though.

pearsonk, 2017-05-17:
Some additional complexity comes with any intent to support more than Logitech keyboards. While some functions such as setting key color and intensity are fairly universal, power-on effects and native effects are likely not as common, therefore not best to be exposed at the highest class in the hierarchy. One way around this could be dynamic_cast checks for implementers, or logic in the implementer based on vendor ID, especially if there is a vendor-specific intermediary class that exposes vendor-specific implementations such as native effects capabilities.

pearsonk, 2017-05-17:
[image]

Here's an illustration of what I was talking about.

MatMoul, 2017-05-17:
This was one of my idea but I didn't have time to do it.
I think it's good direction to be followed for the next step.

aik18, 2017-08-04:
I've had a logitech G510s and G15 keyboard, and I wrote a c++ userspace program to control the leds. Full control over the G510 and G13 keyboards works as well. I could provide the code.

MatMoul, 2017-08-08:
@aik18 Thank you for the proposal, I'm very interrested.

MatMoul, 2019-01-04:
Closing because expired issue

closed 2019-01-04 by MatMoul