MatMoul/g810-led #83: Processing and IO Reduction for setAllKeys

pearsonk, 2017-05-17:
Currently calls to setAllKeys performs an iterative approach to enumerate the keys or regions, then perform a setKeys operation on the final total. The positive of this is that it provides a fairly simplistic way to test each key independently in a single sweep. The trade-off for this is processing time and a lot of USB IO for per-key-rgb devices in particular.

If we would like to improve efficiency here, I would recommend changing the implementation of setAllKeys to simply call setNativeEffect(NativeEffect::Color, NativeEffectPart::all,0x0, <color>) for keyboards that support the native effect, which is far more efficient.

Just a thought, depending on the intent behind setAllKeys. Can be addressed via documentation later.

MatMoul, 2017-05-17:
setAllKeys can not be replaced by setNativeEffect because firstly setAllKeys is part of setting individual keys and setNativeEffect is part of keyboard effect.
And secondly, when you use setAllKeys, you can set other keys in a second time.
But if you use setNativeEffect and you set a key in a second time, you loose the effect,

pearsonk, 2017-05-17:
Good point, I hadn't considered that. Closing this out.

closed 2017-05-17 by pearsonk