MatMoul/g810-led #84: API/Library Documentation

pearsonk, 2017-05-17:
It would be good to have documentation for the library API to inform integrators about the exposed functions and data types. This can also aid contributors to better understand the system, functions, and constants without having to look through the source. There are a lot of documentation systems out there that can auto-generate documentation based on code comments, meaning documentation only needs to be created in one spot. I have used Doxygen in the past and found it quite easy to use for anyone familiar with Javadoc-style document generation.

MatMoul, 2017-05-17:
It would be good to have documentation but that take time and need lot of maintenance...
It's not a priority but we keep in mind.

MatMoul, 2017-05-18:
In the beginning I had to create a wiki but it does not seem very community or I did not understand how it worked ...
Another idea would be to open a branch gh-pages

MatMoul, 2017-06-29:
@pearsonk I just saw you have worked hard...
I'm very busy these days.
You can work on your fork without risk of change from me before the end of summer.
On end of summer, if all are ok, we can make the point and merge your changes.

MatMoul, 2019-01-15:
No news from 2017, I close this issue
@pearsonk you're free to reopen it..

closed 2019-01-15 by MatMoul