MatMoul/g810-led #85: F keys not functioning via Profile

PFCKrutonium, 2017-05-20:
My Current Profile (/etc/g810-led/profile)
```
g logo 000096       # Blue
g indicators ffffff # White
g multimedia 009600 # Green
g fkeys ff7700	    # Orange
g modifiers ff7700  # Orange
g arrows 0000ff     # Red
g numeric 00ff00    # Green
g functions ffffff  # White
g keys ff00ff       # Purple
g gkeys ffffff      # White
k tilde ff0000      # Red
k W 0000ff          # Blue {
k A 0000ff
k S 0000ff
k D 0000ff          # }         
k 1 ff0000          # Red {
k 2 ff0000
k 3 ff0000
k 4 ff0000
k 5 ff0000
k 6 ff0000
k 7 ff0000
k 8 ff0000
k 9 ff0000
k 0 ff0000          # }
c                   # Commit changes
```
Issue: F keys do not light up when this profile is used. Manually changing them via commands ie `g810-led -g fkeys ff7700` works, but `g810-led -pp < /etc/g810-led/profile` does not. New Issue Report created to cover the remaining issue from #81 .

MatMoul, 2017-05-20:
Ok I just found a problem...
try with g810-led -p /etc/g810-led/profile
Seem a problem in the pipe method

MatMoul, 2017-05-20:
No, I have writed to speed...
But try to delete your fkeys line et rewrite it you have a bad char

PFCKrutonium, 2017-05-20:
Whelp that fixed it. I wonder how that line ended up messed up in the first place.

MatMoul, 2017-05-20:
It was rather bad as a bug ...
But it allowed me to correct my sendDataInternal function
Ready for the next version

PFCKrutonium, 2017-05-20:
Hey, sometimes the best bugs are the ones where you discover some input just breaks things :+1:

closed 2017-05-20 by MatMoul