MatMoul/g810-led #88: g810-led: name idea.

kraaijmakers, 2017-05-21:
Hello g810-led community, awesome work you have done already! :) 

So the idea for naming, I was thinking about the name **logiled**. Unless the software should be able to control multiple vendors their keyboards I think logiled will suffice for this piece of software.
Since people are already trying to add their mouse to the software and stuff, I think we should leave out any model number out of the name. 

Thanks for reading, and happy coding!

P.s.: This is my first time trying to 'contribute' somehow to something. So I hope this will work and is done properly. If not, let me know!

MatMoul, 2017-05-22:
Thank for your contribution, now I wait if another manufacturer can be supported

MatMoul, 2017-05-22:
This issue can be related to #29

kraaijmakers, 2017-05-22:
if multiple keyboard vendors will be supported it could be named openkbdled or something like that :)

closed 2017-05-22 by kraaijmakers

MatMoul, 2017-05-22:
Thanks, I take time to choose an other name because renaming a project has large consequences.
But all ideas are welcome.

ShayBox, 2019-01-19:
I'm not sure what the name should be, but it really should be changed as the name is pretty confusing, it took me 6 times to figure out that this repo works with a g910, i thought I had to use the outdated fork...

MatMoul, 2019-01-22:
@ShayBox Some distributions have now added this project to their repository and renaming it becomes complicated ...

ShayBox, 2019-01-22:
Not impossible, The only real problem would be links to files on the repo
and github continues to provide working links after rename for awhile.

MatMoul, 2019-01-22:
Just search on Github...

https://github.com/topics/logitech-gaming
https://github.com/topics/logitech

https://github.com/topics/g410
https://github.com/topics/g610
https://github.com/topics/g810
https://github.com/topics/g910
https://github.com/topics/g213
https://github.com/topics/gpro
https://github.com/topics/g413
https://github.com/topics/g512
https://github.com/topics/g513

ShayBox, 2019-01-22:
That's not really... related to this but that's cool, it would still be
great if the project name wasn't so misleading, although it does what it
says, it does a lot more than it says.