MatMoul/g810-led #89: g410 keyboard crashing/restarting

letoram, 2017-05-21:
Hi,
I'm trying to use g810-led for a custom project (https://arcan-fe.com/2017/05/21/playing-with-leds/)  keeping the program running and being continuously fed changes through a pipe. At random times,
 the commit() function fails and the keyboard does a full disconnect/connect "splash- color" cycle - or setAllKeys fail to set all keys.

Replaying the same input sequence rarely reproduces the issue, so there's some timing / buffering component involved that I am not seeing.

The patched/modified area is here: https://github.com/letoram/aledctrl/blob/master/g810/src/main.cpp#L107

here is a dump of an usbmon 1u dump of the device (latest firmware) just before a crash:
https://github.com/letoram/aledctrl/commit/a01b7b9e59c5fbf49e92a6e978b10da430b3da80

any hints/suggestions, or am i misinterpreting how the internal APIs work?

MatMoul, 2017-05-22:
I just maked a new version 0.2.5 that can be resolve this issue (not sure).
Remark : This version has a bug for multiple keyboard #90 but you can test with it.

I have tryed to build your project and launched g810-led --aled -
But how I use it after ?

letoram, 2017-05-22:
added a stress-test that can trigger the issue:
https://github.com/letoram/aledctrl/tree/master/testgen

can be run like:

    ./test | g810-led --aled -

observation:
if I do nothing, the test ran without a problem - but if I started typing fast on the keyboard at the same time, the crash came quickly. 

it seems like 0.2.5 resolved it though!

letoram, 2017-05-22:
been running the patch without issue for the last ~3 hours now so think it can be considered fixed. Thanks!

closed 2017-05-22 by MatMoul