MatMoul/g810-led #90: Recent commit breaks multi-keyboard or targeted commands

pearsonk, 2017-05-22:
Commit ID d6d976322e074e470059eed6dce7ff53dcb11709 breaks the support for multiple keyboards or targeted device by only sending the first data chunk to the target device, then reverting to the "first supported device" for subsequent calls.

Tested case: a G810 plugged in first, then a G410 plugged in, intent to send commands to the G410.
Running './bin/g810-led -dp c330 -a FF0000' results in all the keys present on the G410 being set to red, but erroneously on the G810 device. This is because the target is initially mapped via a targeted open(uint16_t,uint16_t,string). The first run through sendDataInternal runs fine, but closes the device. Subsequent calls to open() will not return early because m_isOpen is now false, so it will re-open the first found device, the G810.

MatMoul, 2017-05-22:
Thank for your remark.
Your right, I missed this part of the code...
I try to update this as soon as possible

MatMoul, 2017-05-22:
I've just pushed a branch named "patch_multiple_keyboard_bug".
Can you test it ?

This is a fast correction but I've just see that the open function reset the currentDevice.model and restart all the code detection when I call it a second time.
This can be improved.

pearsonk, 2017-05-26:
This fix works

MatMoul, 2017-05-26:
Thanks, I commited this change in the develop branch.

closed 2017-05-26 by MatMoul