MatMoul/g810-led #92: G810 UK Layout

FruityFaiz, 2017-05-30:
Hey, for some reason I cannot get it to work with my G810. When I execute `g810-led --list-keyboards`, it lists my keyboard but when executing anything else I get 

> Matching or compatible device not found !

When I installed it, it also changed my keyboard's colours so it seems to be working fine. 

I'm on Arch Linux, if you need any dumps etc just let me know.

Thank you!

MatMoul, 2017-05-30:
If the --list-keyboard command work and when your have installed g810-led your keys have changed color, your keyboard are supported.
Do you have tryed g810-led -a ff0000?

FruityFaiz, 2017-05-30:
Hey, same problem when executing `g810-led -a ff0000`.
It seems to detect the keyboard(right IDs) when Keyboard::open is called but hid_open() seems to be returning a NULL pointer for some reason.

MatMoul, 2017-05-30:
Very strange, this is the same internal code that is called after installation and it works (dump not required).

Need more investigation...
Without destroying your computer, try with a clean install of ArchLinux on a second disk or a vm.
If it works, you must have a software that enter in conflict with g810-led.

FruityFaiz, 2017-05-30:
Seems to be my system or hidapi, running the code beneath prints

> Failed to open device.

`#include <iostream>
#include <hidapi/hidapi.h>

int main()
{
	if (hid_init() < 0)
		std::cout << "Failed to initialise." << std::endl;	

	auto dev = hid_open(1133, 49975, NULL);

	if (dev == 0) {
		hid_exit();
		std::cout << "Failed to open device." << std::endl;
	}

	return 0;
}
`

I will get back to you with what I find out. Thanks!

pearsonk, 2017-05-30:
Try running your code as root (difference between the installed udev base/profile and normal user). This may indicate a permissions or hidapi problem.

You can also get more information if you add prior to your hid_exit call of your test application:
`std::wcout << hid_error(dev) << std::endl;`
Requires including wstring.

FruityFaiz, 2017-05-30:
Oh gosh! Running as root worked. Thanks. 

And extra information isn't printed when adding that just before the exit call.

MatMoul, 2017-05-30:
Very strange, the udev rules contain MODE="666" that allow all users to use it.
Do you use the aur package or make and make install?

FruityFaiz, 2017-05-30:
I used both, same problem.

MatMoul, 2017-05-30:
Ok, I recommand you to use the aur package, with it, you receive updates...

It seem that is a udev problem, but normally, it is include in systemd.

FruityFaiz, 2017-05-31:
Just started working after a restart...

closed 2017-05-31 by FruityFaiz