MatMoul/g810-led #93: G810 - Rainbow affect G logo

FruityFaiz, 2017-06-03:
Hey, I was wondering if it would be possible to have the G logo sync with the rainbow affect. WIth the current logitech gaming software, the G logo also changes colours.

Using the following command: `g810-led -fx hwave keys 0a` does not affect the G logo or indicators.

Thank you!

MatMoul, 2017-06-04:
I have tried to do it, but I can't sync the logo animation with the keys animation...
You can try by calling g810-led -fx cycle logo 0a... (this is the method extracted from the dump)
Even if you change the speed, the logo animation loses sync...

FruityFaiz, 2017-06-04:
It's fine, would be nice but not a problem much, I can live.

If you ever do decide to look into it, just to let you know the colour changes from red to green rapid, instead of going yellow then green. I think that might be your issue.

MatMoul, 2017-06-08:
I would try to look into it if I have time.
But for now, I work on other features that is a priority for me.

FruityFaiz, 2017-06-16:
Ye no problem doesn't matter. Just let me know if you get it fixed :+1:

A1rPun, 2017-10-19:
I'm very happy with this command which I extracted from this issue 😄 
```
g810-led -fx hwave keys ff && g810-led -fx cycle logo ff
```
The desync isn't that terrible

MatMoul, 2018-10-23:
Can I close this issue ?

A1rPun, 2018-10-24:
@MatMoul I would love this feature being implemented because I always use `g810-led` in combination with the rainbow effect! I assume you looked into this so if it is not possible there is no choice other than closing this issue.

MatMoul, 2018-10-24:
Thank's for your return and sorry that I can make anything more

closed 2018-10-24 by MatMoul