MatMoul/g810-led #94: Compatible with profiles from LGS?

benutzer74, 2017-06-07:
When I save a profile with Logitech Gaming Software (LGS) in Windows7, it seems to create an xml format file. Can your tool load profiles created with LGS? When I look at your sample profile they look different from LGS´ profiles? I only want to assign different static colors to different G810 keys, no fancy effects. Thanks.

MatMoul, 2017-06-07:
You'r right, my profile files aren't the same as Logitech.
If you open a Logitech file, it's very hard to parse her schema and I have other priorities at this time.
But it's very easy to make your own profile file with my format.

benutzer74, 2017-06-07:
Since I just want to assign static colors to keys without fancy effects, I think g810-led-python would do the job just fine? Thanks!

MatMoul, 2017-06-08:
Thanks for your offer...
But it's very difficult to send a package to the milky way's galaxy :-p

closed 2017-06-08 by MatMoul