MatMoul/g810-led #95: is a reactive mode possible

xdotax, 2017-06-29:
Hello and thank you for your hard work on this project. Is it possible to set a reactive mode, whereby the keys that are pressed change color and then the color fades back to the original color over a period of time? thank you again.

MatMoul, 2017-06-29:
At this time, it's not possible because g810-led don't reside in memory and can't track key pressed.
I love these types of effects but they include side effects.
Anyone near you can see your passwords when you type it.
And for working I need to make a keylogger. But I don't love this idea.
Perhaps in the future...

snkiz, 2017-08-30:
There are benefits. I was was going to ask if g810-led can read a keys current colour somehow because I wanted to write a script to light up the caps/scroll/num lock key when they are active then return them to the previous colour when not. I have an idea for a kebal space program mod to toggle colours on the keyboard in response to game events as well. 
That would need to be written with mono libs, I poked at the logigsk api docs they look like they have C# bindings. I don't really know what to do with this....

But For the lock key problem. I can get the key sate with pretty easily with xset, or there's this python script stuff I found:
https://www.linuxquestions.org/questions/linux-general-1/how-do-i-use-the-python-module-keyboardleds-to-determine-caps-lock-state-4175554587/
He mentions a python module called keyboardleds I don't know much about python yet. I do know that relying on X based commands is only going to last as long as the propriety drivers lack wayland support. So there's that. I guess I'm asking where do I go from here?

MatMoul, 2017-08-30:
@snkiz I don't have maked the reverse engeneer for reading the key state.
Time is missing but I look to make a user driver which could be more flexible...

pyrocks66, 2017-09-15:
@snkiz > I was looking to save and restore key colors for a python script too. I found this project can read current key colours: https://github.com/spectras/keyleds/

snkiz, 2017-09-15:
Now Thats intresting. its in C, has py bindings and a strange dependency on Qt Core. I'll try it But @MatMoul  You two should get in touch, you have groups. That project doesn't mention them.

MatMoul, 2017-09-27:
@snkiz Thanks, I also saw this project, it seems interesting...
I will study what I will do for the next

spectras, 2017-10-05:
Hello!

I am the developer of keyleds. I don't really know where to go from here… the initial impulses of g810-led and keyleds seem similar, but the scope is quite different. Keyleds is a user-session service, its aim is to provide the same features as Logitech software for windows, and then some more.

I guess that answers some of the questions:
@snkiz the Qt dependency is for the main event loop and for the DBus interface: keyleds exposes the ability to swap effects and send events through DBus. Keyleds does feature key groups, only you define them in the configuration file (eg `arrows: [left, right, up, down]`), it doesn't come with predefined ones, though the sample configuration has sensible examples.

What from there, I don't know. You seem to be very busy atm @MatMoul and I don't know your own perspectives on g810-led?

MatMoul, 2017-10-05:
@spectras 
Yes, I'm verry busy these days...
But I've see your project and it's very interesting...
I'll comming back to you in 2 or 3 month, I need to work on an other project these next month

closed 2018-09-13 by MatMoul