MatMoul/g810-led #98: Wayland Support

arcadian09, 2017-07-24:
On Fedora 26 with Wayland and cannot compile/install.

Got the following errors:
`Jul 24 08:43:05 lt-oleg-t450s kernel: input: Logitech Gaming Keyboard G610 as /devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.1/0003:046D:C338.0050/input/input85
Jul 24 08:43:05 lt-oleg-t450s kernel: hid-generic 0003:046D:C338.0050: input,hiddev0,hidraw6: USB HID v1.11 Keyboard [Logitech Gaming Keyboard G610] on usb-0000:00:14.0-3/input1
Jul 24 08:43:05 lt-oleg-t450s systemd-udevd[26056]: Process '/usr/bin/g610-led -p /etc/g810-led/profile' failed with exit code 2.
Jul 24 08:43:05 lt-oleg-t450s systemd-udevd[26053]: Process '/usr/bin/g610-led -p /etc/g810-led/profile' failed with exit code 2.
Jul 24 08:43:05 lt-oleg-t450s gnome-shell[1982]: Could not open device /dev/input/event8: Could not get device info for path /dev/input/event8: No such file or directory
`

MatMoul, 2017-07-24:
Not sure that a Wayland problem, g810-led work on concole...

What the result of g810-led --list-keyboards ?
And is g810 -a 00ff00 working?

MatMoul, 2017-07-24:
I don't have a PC that support Wayland for testing.
But I don't use xinit...
I use hidapi that is dependent of libusb. These libs don't require Xorg or Wayland or xinit or libinput.
On my PC I use libinput with Xorg and that work.
I suspect that you have a make problem because you have a segmentation fault...

arcadian09, 2017-07-24:
ok, thats a nice thought. use it with hidapi.

After installaing hidapi and hidapi-devel. I compiled the bin: `make bin`

Got this resolt

 **./g810-led --list-keyboards**
`Device: Logitech - Gaming Keyboard G610`
`	Vendor ID: 046d`
`	Product ID: c338`
`	Serial Number: 0760366E3738`

Looks like I am good :). 
Thanks

MatMoul, 2017-07-24:
Good work!
see you

closed 2017-07-24 by MatMoul