MatMoul/g810-led #99: G413 keyboard backlighting control

vooon, 2017-07-24:
Logitech's software sets "breathing" effect for startup and don't change it! Very annoying.

I tried several combinations, and found that this commad actually change brightness:

    sudo ./bin/g810-led -dv 046d -dp c33a -tuk 3 -fx color all 800000

I'm not sure that dumps you need to add support, but at least i fixed my problem.
Thank you!

MatMoul, 2017-07-24:
Thank's for your return.
The g413 is not supported now.
But with your command line, I can add it as supported keyboard.
Just need a little time to make a new version...

MatMoul, 2017-07-26:
I work on the g413 and the g213 don't need commit.
can you test th next cmd :
g810-led -dv 046d -dp c33a -tuk 3 -an 600000
And return me if this cmd change your keyboard.
If not, try this next cmd :
g810-led -dv 046d -dp c33a -tuk 3 -c
And return me if the second cmd are required

MatMoul, 2017-07-26:
@vooon I've just pushed a new branch named g413, can you test it ?
`git clone https://github.com/MatMoul/g810-led.git`
`cd g810-led`
`git checkout g413`
`make bin`
`sudo make install`
With this you can use g413-led

vooon, 2017-07-28:
Sorry i was busy.
First command: no changes.
Second: no changes.

And i'm not sure that this kb don't use commit, since it remember effect (uses it on reset).
Also i don't reboot to windows yet.

vooon, 2017-07-28:
```
$ sudo ./bin/g810-led --print-device
Device: Logitech - G413 Carbon Mechanical Gaming Keyboard
	Vendor ID: 046d
	Product ID: c33a
	Serial Number: 106C366E3933
vovan@trurl:~/src/misc/g810-led$ sudo ./bin/g810-led -fx breathing all ff0000 1
zsh: exit 1     sudo ./bin/g810-led -fx breathing all ff0000 1
vovan@trurl:~/src/misc/g810-led$ sudo ./bin/g810-led -fx color all 800000 1
zsh: exit 1     sudo ./bin/g810-led -fx color all 800000 1
```

Both `-fx` works.

MatMoul, 2017-07-28:
Ok, it seem that is not the -tuk 3, -fx share code for many keyboards...
When you have time, try with -a.
g810-led -dv 046d -dp c33a -tuk 1 -a 600000
g810-led -dv 046d -dp c33a -tuk 2 -a 600000
g810-led -dv 046d -dp c33a -tuk 3 -a 600000
And return me if one of these commands update your keyboard.
Thank you

vooon, 2017-07-29:
None of that commands changes anything.
Also seems that `-fx color all 800000` remembers as startup default (keep that setting between reboots).

MatMoul, 2017-07-30:
No luck, if all three commands have no effect, new dumps are needed

vooon, 2017-07-30:
How to do that?

MatMoul, 2017-07-30:
It's not very difficult, on a Windows, install and start Wireshark.
Launch the Logitech software and when your ready to apply a config, start the capture of the usb port with Wireshark.
You can look the other dumps at https://github.com/MatMoul/g810-led-resources/tree/master/dumps.
Try to make one dump per opération.
When ready you can send me the files in a zip here or make a PR on https://github.com/MatMoul/g810-led-resources.

vooon, 2017-08-01:
I done some captures of composite device when i interact with UI.
There are no many options, only brightness slider, breathing effect checkbox and breathing speed slider.

What config you talking about?

MatMoul, 2017-08-01:
Thanks for the dumps...
I just read them and it seem that this keyboard has only color effect and breathing effect.
The two next cmd need working on current release :
g810-led -dv 046d -dp c33a -tuk 3 -fx color all 80
g810-led -dv 046d -dp c33a -tuk 3 -fx breathing all 80 0a
80 = intensity
0a = speed
Can you confirm ?

vooon, 2017-08-01:
Both working, but returns 1 exit status.
```
vovan@trurl:~/src/misc/g810-led$ sudo ./bin/g810-led -dv 046d -dp c33a -tuk 3 -fx breathing all 80 0a 
zsh: exit 1     sudo ./bin/g810-led -dv 046d -dp c33a -tuk 3 -fx breathing all 80 0a
vovan@trurl:~/src/misc/g810-led$ sudo ./bin/g810-led -dv 046d -dp c33a -tuk 3 -fx color all 80       
zsh: exit 1     sudo ./bin/g810-led -dv 046d -dp c33a -tuk 3 -fx color all 80
```

MatMoul, 2017-08-02:
I just pushed a new g413 branch.
With it, you can also use the -a arg to set all keys.
The exit 1 is a bug that I have corrected.

vooon, 2017-08-02:
`-a` works.

MatMoul, 2017-08-02:
Thank you
Now I can merge changes and make a new release.

MatMoul, 2017-08-09:
@vooon Thanks for your contribution...
I just pushed the new realease with g413 support.